ARM-software/tf-issues #491: How to create ROTPK hash to feed plat_get_rotpk_info()

masahir0y, 2017-06-09:
Platforms that use TBB must store the hash of ROTPK hash
(or skip ROT verification completely by setting ROTPK_NOT_DEPLOYED flag)

I know tools/cert_create/cert_create can generate certificates files,
but does it provide a way to generate ROTPK hash for plat_get_rotpk_info()?

soby-mathew, 2017-06-09:
@masahir0y, cert_create tool does not provide to generate ROTPK hash although the rotpk to be used can be specified via the `--rot-key` option. For ARM reference platforms, if the ARM_ROTPK_LOCATION=devel_rsa is specified, then the public key in [arm_rotprivk_rsa.pem](https://github.com/ARM-software/arm-trusted-firmware/blob/master/plat/arm/board/common/rotpk/arm_rotprivk_rsa.pem) used as the rotpk.

I may not be understanding your query correctly, but why do you need to generate ROTPK hash? For development purposes you can specify a ROTPK as mentioned above or use the ROTPK_NOT_DEPLOYED flag.

If you are referring to generating `ASN.1 DER` format from raw ROTPK hash (as specified [here](https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/porting-guide.md#function-plat_get_rotpk_info)), please refer to the [plat_get_rotpk_info()](https://github.com/ARM-software/arm-trusted-firmware/blob/master/plat/arm/board/common/board_arm_trusted_boot.c#L56) implementation for ARM reference platforms.

geesun, 2017-06-12:
@masahir0y，I use the following way generate the ROTPK. 
1. Generate the private and public key which will use in build command option ROT_KEY=rotprivk_rsa.pem
`openssl genrsa -out rotprivk_rsa.pem 2048`
2. Extract Public from private key
`openssl rsa -in rotprivk_rsa.pem -inform PEM -pubout -outform DER -out rotpk_rsa.der`
3. Generate 256 bit SHA-256 HASH of public key
`openssl rsa -pubin -inform DER -in rotpk_rsa.der -outform der | openssl dgst -sha256 -binary > rotpk_rsa_sha256.bin`
4. Program public rotpk_rsa_sha256.bin to OTP as  ROTPK

masahir0y, 2017-06-14:
@soby-mathew 
Thanks for the pointer.  I was so surprised when I realized that ARM put ROT key in plat/arm/board/common/rotpk/.  Moreover, .der and .bin can be derived from arm_rotprivk_rsa.pem by using the command suggested by @geesun .

Then, finally, I understood the mysterious

    #if (ARM_ROTPK_LOCATION_ID == ARM_ROTPK_DEVEL_RSA_ID)
    static const unsigned char arm_devel_rotpk_hash[] =	\
    		"\xB0\xF3\x82\x09\x12\x97\xD8\x3A"	\
    		"\x37\x7A\x72\x47\x1B\xEC\x32\x73"	\
    		"\xE9\x92\x32\xE2\x49\x59\xF6\x5E"	\
    		"\x8B\x4A\x4A\x46\xD8\x22\x9A\xDA";
    #endif

This is hard-coding plat/arm/board/common/rotpk/arm_rotpk_rsa_sha256.bin instead of embedding it by Makefile.  Jesus christ!

masahir0y, 2017-06-14:
@geesun 
Thanks, this is the answer I was searching for.

I think, basically there are two approach for cert_create.

[1] Let cert_create automatically generate needed keys, then save the keys. 
    (this is possible by setting `SAVE_KEYS`)
[2] Like you suggested, create the rot key by openssl command, then feed it to cert_create.

At first, I was trying [1] (so, I thought it would be reasonable if cert_create supports creating the sha256  of rotpk).

[2] seems straightforward and I can live with it.  I can live with this approach.  Thanks.

danh-arm, 2017-06-19:
Seems like this questions is answered...

closed 2017-06-19 by danh-arm