ARM-software/tf-issues #492: Scenario to receive a FIQ in ATF without EL3 and S-EL1 interrupts in the system

chinnabha, 2017-06-09:
Hi @Soby-mathew,

Please help to know if there is any scenario wherein ATF can receive a FIQ under following conditions:
1. No EL3 (Grp0) and S_EL1 (S-Grp1) interrupts registered in the system.
2. No secure applications running in S-EL0.

danh-arm, 2017-06-12:
Is this a hypothetical question or are you seeing a real problem? Are you referring to a GICv3 system (since you use Grp0 and S-Grp1 terminology)? In a GICv3 system, FIQ is used to signal NS-Grp1 interrupts while executing in secure state. If your system has a Secure OS or application that executes while NS-Grp1 interrupts are active, that could be the scenario you are seeing.

chinnabha, 2017-06-12:
Yes, it was a hypothetical question for a system with GICv3 interrupt controller. 

Ok, let me remove the Secure OS and secure applications too from the system (in addition to above 2 constraints). In this scenario can there be a possibility for Secure Monitor to see a fiq?

danh-arm, 2017-06-12:
> In this scenario can there be a possibility for Secure Monitor to see a fiq?

I don't think so. Interrupts are always masked while executing in EL3. By default, when executing at Exception levels below EL3, physical FIQ interrupts are not taken to EL3, and in your scenario, this is not changed.

Note, the RichOS could register itself as a target for FIQs, but that will not cause EL3 to receive it.

chinnabha, 2017-06-12:
Thanks @danh-arm. Agree, this is aligning with my understanding.

closed 2017-06-12 by chinnabha