ARM-software/tf-issues #493: SCR_EL3 not saved in handle_interrupt_exception( ) but restored in el3_exit( )

chinnabha, 2017-06-12:
Hi,

May I know why SCR_EL3 is not stored into the SP_EL3 stack offset (CTX_EL3STATE_OFFSET + CTX_SCR_EL3) during the IRQ/FIQ handling via the macro handle_interrupt_exception( ). But it is restored in el3_exit( ) macro. 

This seems to be a bug. Please advise.

soby-mathew, 2017-06-12:
The value of SCR_EL3 in the for a given security state in the cpu context and the actual register can guarenteed to be the same as:
*  SCR_EL3 can only be modified by the software entity running at EL3. Hence between the exit to lower exception level and return to EL3 via interrupt, this register is not modified.
* The generic code (except the entry and exit code) only ever writes to the copy of SCR in the context.

So as long as the value of SCR in the context is written to the register prior to exit to lower exception level, we should be fine I think.

chinnabha, 2017-06-12:
SCR_EL3 is stored and restored during SMC handling from the stack pointed to by SP_EL3. Whereas it is restored but not saved during irq/fiq handling in EL3.

I am not sure what is stopping us in storing and restoring it in the irq/fiq handler path? It hardly saves us with one instruction.

soby-mathew, 2017-06-12:
Yes, you are right that the SMC handling saves the register in the cpu context on entry to EL3 but, unless I am missing something, it is redundant (because the register couldn't have changed between the previous exit to lower exception level and the entry to EL3 now). 

If some EL3 code is doing something outside the given frameworks (like writing directly to scr_el3 and then doing an eret directly instead of el3_exit)  then perhaps it is useful, but I dont see value otherwise.

chinnabha, 2017-06-12:
Sure @soby-mathew. I am fine if you are convinced its unnecessary.

closed 2017-06-12 by chinnabha