ARM-software/tf-issues #497: v1.4 release testing started

danh-arm, 2017-06-29:
We've started release testing for the TF v1.4 release. To facilitate this we've created a temporary release branch:

https://github.com/ARM-software/arm-trusted-firmware/tree/for-v1.4

All remaining v1.4 content will be placed on this branch. Non-v1.4 content will be merged to `integration` as usual, but not caught up with master. Once the v1.4 release is complete (hopefully within a week), the `master` branch will be fast-forwarded to `for-v1.4` and any content that has built up on `integration` will be rebased.

vwadekar, 2017-06-30:
Any plans to maintain the release branch after the release? That way one can just take v1.4 until the next release and doesnt have to worry about master.

danh-arm, 2017-07-04:
No, all changes after the release will only be made on master. Maintaining release branches is too costly.

closed 2017-08-14 by danh-arm