ARM-software/tf-issues #499: cert_create tool broken for existing ATF implementations

sbranden, 2017-07-04:
@soby-mathew
@danh-arm 
The follow commit breaks cert_create for existing platforms:
cert_create: Use RSASSA-PSS signature scheme for certificates

It's fine added support for RSASSA-PSS signature scheme but we need support for the existing certificate creation.

Also, the commit states "This is compliant with RSA PKCS_2_1 standard as mandated by TBBR".  Could you please point me at the TBBR document mandating this?

danh-arm, 2017-07-05:
Hi @sbranden 

Could you elaborate why you need cert_create support for the previous PKCS1_5 standard certificate signature scheme? Use of this signature scheme was a mistake since it is not compliant with the TBBR spec. Therefore we deprecated it in favour of the TBBR compliant PKCS_2_1 standard scheme. The device side code continues to support both schemes for compatibility with old tools, but we don't think it's generally a good idea for new certificates to be generated using the old scheme.

The only legitimate use-case for using the old scheme I can think of is to support unpatchable BL1 code that only understands the old scheme. Is that your case? If so, then perhaps any new BL2 for such platforms can be generated using an old cert_create version, while the rest of the chain of trust uses the latest cert_create (and signature scheme)?

The TBBR document (ARM DEN0006C-1) is available to ARM partners via your account manager.

Regards

Dan.

sbranden, 2017-07-05:
Hi Dan,

Yes, BL1 is in ROM.  We need cert_create to work with those SoCs.

danh-arm, 2017-07-06:
OK @sbranden. We can add a command line option to cert_create to use the old signature scheme. The current behaviour would remain the default, though.

soby-mathew, 2017-07-06:
Internal ref: https://jira.arm.com/browse/GENFW-1959

sbranden, 2017-07-06:
Yes, a command line option would be appropriate to select the necessary signature scheme.
Perhaps in the future another signature scheme will be used so having option able to select 1.5 or 2.1 scheme seems appropriate.  If no scheme selected then, yes, default to 2.1.

closed 2017-09-07 by davidcunado-arm