ARM-software/tf-issues #500: about bl1 securty memory

yym-hjr, 2017-07-05:
hello:
Why bl1 initialization memory function is empty? Bl1 is the beginning of the operation, I would like to ask where to run?
Looking forward to your reply!

danh-arm, 2017-07-05:
Hi

I guess you're talking about the platform porting function, `platform_mem_init()`? This function allows the platform to perform any essential memory initialization before BL1 accesses memory. On all upstream platforms, BL1 memory (for example on-chip SRAM) is available to use immediately without any explicit initialization, therefore this function is empty. Note that later firmware may need to do DRAM initialization before using that memory. This is very platform specific, though. On ARM platforms, this is done as part of loading SCP_BL2. On other platforms, this might be done within `bl2_platform_setup()` or `bl31_platform_setup()).

> I would like to ask where to run?

Are you asking what memory to use to execute BL1, or where the memory initialization is done, or something else? I think you need to elaborate on your question.

Regards

Dan

yym-hjr, 2017-07-06:
@danh-arm 
I would like to ask what memory to use to execute BL1, , I think, BL1 is running in the configuration of the safe memory above, rather than simply running on the most basic sram. However, I did not see the configuration for secure memory.
Looking forward to your reply!

Regards

Yi

danh-arm, 2017-07-06:
Hi @yym-hjr. Each platform can choose what type of memory to use for BL1, depending on its requirements. Typically we expect BL1 code and read-only data to be placed in ROM. Each platform can configure this by defining `BL1_RO_BASE` and `BL1_RO_LIMIT`. Typically we expect BL1 read-write data to be placed in on-chip SRAM. The platform can configure this by defining `BL1_RW_BASE` and `BL1_RW_LIMIT`. See the in-source [porting guide](https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/porting-guide.rst) for more info.

soby-mathew, 2017-07-07:
Hi @yym-hjr,
The Trusted on-chip SRAM is `Trusted` out of reset.  So usually configuration is not required to make SRAM `Trusted` if that is what you are referring to. BL1 should only run from `Trusted SRAM`.

Best Regards
Soby Mathew

yym-hjr, 2017-07-08:
@soby-mathew   thank you

closed 2017-07-08 by yym-hjr