ARM-software/tf-issues #503: qemu target doesn't build for 1.4 with gcc-7

nullr0ute, 2017-07-08:
Other targets build fine such as juno rk3399 rk3368 rk3328 hikey hikey960. Qemu gives the following error "lib/psci/psci_common.c:198:27: error: array subscript is above array bounds [-Werror=array-bounds]"

Full output for particular line:
<pre>
gcc -DDEBUG=0 -DNDEBUG -DSTACK_PROTECTOR_ENABLED=0 -DBL32_RAM_LOCATION_ID=SEC_DRAM_ID -DSKIP_A57_L1_FLUSH_PWR_DWN=0 -DA53_DISABLE_NON_TEMPORAL_HINT=1 -DA57_DISABLE_NON_TEMPORAL_HINT=1 -DERRATA_A53_826319=0 -DERRATA_A53_835769=0 -DERRATA_A53_836870=0 -DERRATA_A53_843419=0 -DERRATA_A53_855873=0 -DERRATA_A57_806969=0 -DERRATA_A57_813419=0 -DERRATA_A57_813420=0 -DERRATA_A57_826974=0 -DERRATA_A57_826977=0 -DERRATA_A57_828024=0 -DERRATA_A57_829520=0 -DERRATA_A57_833471=0 -DCRASH_REPORTING=0 -DARM_CCI_PRODUCT_ID=400 -DARM_ARCH_MAJOR=8 -DARM_ARCH_MINOR=0 -DARM_GIC_ARCH=2 -DCOLD_BOOT_SINGLE_CPU=0 -DCTX_INCLUDE_AARCH32_REGS=1 -DCTX_INCLUDE_FPREGS=0 -DENABLE_ASSERTIONS=0 -DENABLE_PLAT_COMPAT=0 -DENABLE_PMF=0 -DENABLE_PSCI_STAT=0 -DENABLE_RUNTIME_INSTRUMENTATION=0 -DERROR_DEPRECATED=0 -DHW_ASSISTED_COHERENCY=0 -DLOAD_IMAGE_V2=0 -DLOG_LEVEL=20 -DNS_TIMER_SWITCH=0 -DPL011_GENERIC_UART=0 -DPLAT_qemu -DPROGRAMMABLE_RESET_ADDRESS=0 -DPSCI_EXTENDED_STATE_ID=0 -DRESET_TO_BL31=0 -DSEPARATE_CODE_AND_RODATA=0 -DSPD_none -DSPIN_ON_BL1_EXIT=0 -DTRUSTED_BOARD_BOOT=0 -DUSE_COHERENT_MEM=1 -DUSE_TBBR_DEFS=0 -DWARMBOOT_ENABLE_DCACHE_EARLY=0 -DENABLE_SPE_FOR_LOWER_ELS=0 -DAARCH64 -Iinclude/lib/stdlib -Iinclude/lib/stdlib/sys -Iinclude/bl1 -Iinclude/bl31 -Iinclude/common -Iinclude/common/aarch64 -Iinclude/drivers -Iinclude/drivers/arm -Iinclude/drivers/auth -Iinclude/drivers/io -Iinclude/drivers/ti/uart -Iinclude/lib -Iinclude/lib/aarch64 -Iinclude/lib/cpus -Iinclude/lib/cpus/aarch64 -Iinclude/lib/el3_runtime -Iinclude/lib/el3_runtime/aarch64 -Iinclude/lib/pmf -Iinclude/lib/psci -Iinclude/lib/xlat_tables -Iinclude/plat/common -Iinclude/services -Iinclude/plat/arm/common/ -Iinclude/plat/arm/common/aarch64/ -Iplat/qemu/include -Iinclude/common/tbbr  -Iinclude/tools_share -Iinclude/lib/libfdt -nostdinc -Wmissing-include-dirs -Werror -march=armv8-a -mgeneral-regs-only -mstrict-align -ffreestanding -fno-builtin -Wall -std=gnu99 -Os -ffunction-sections -fdata-sections -Wno-error=deprecated-declarations  -DIMAGE_BL31 -Wp,-MD,./build/qemu/release/bl31/psci_common.d -MT build/qemu/release/bl31/psci_common.o -MP -c lib/psci/psci_common.c -o build/qemu/release/bl31/psci_common.o
lib/psci/psci_common.c: In function 'psci_do_state_coordination':
lib/psci/psci_common.c:198:27: error: array subscript is above array bounds [-Werror=array-bounds]
  psci_req_local_pwr_states[pwrlvl - 1][cpu_idx] = req_pwr_state;
  ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~
cc1: all warnings being treated as errors
</pre>

dp-arm, 2017-07-10:
the fvp platform is also affected by this

dp-arm, 2017-07-11:
does not appear to be a bug in TF, more like a false positive in the compiler

danh-arm, 2018-02-22:
Closing as this should have been addressed by https://github.com/ARM-software/arm-trusted-firmware/pull/515

closed 2018-02-22 by danh-arm