ARM-software/tf-issues #505: Query regarding loading multiple Trusted OS support in ATF

b49020, 2017-07-18:
I am new to ATF, so I have following basic query.
Does it seems to be feasible idea to load multiple Trusted OS in ATF assigning specific owning entity number (OEN) to each trusted OS as per SMC calling conventions?

soby-mathew, 2017-07-19:
@b49020,
The Trusted OS runs at S-EL1 and usually the Trusted OS assumes full owner ship of the EL. So currently it is not possible to run more than 1 Trusted OS at S-EL1.

vchong, 2017-07-19:
I'm rather new too.
Is it possible then with multiple cores?

danh-arm, 2017-07-19:
Not without a great deal of effort. Some higher level code would need to manage the multiple S-EL1 contexts, and each Trusted OS would need to convert to using some newly introduced APIs to ensure they play nicely with each other. The trusted boot flow would also be greatly impacted. This may be more feasible in future architecture revisions.

b49020, 2017-07-20:
I did had a look at OP-TEE OS (S-EL1). There they had hard-coded values of OEN numbers as follows:
#define OPTEE_SMC_OWNER_TRUSTED_APP	48
#define OPTEE_SMC_OWNER_TRUSTED_OS	50
#define OPTEE_SMC_OWNER_TRUSTED_OS_OPTEED 62
#define OPTEE_SMC_OWNER_TRUSTED_OS_API	63

Shouldn't it be the responsibility of ATF to provide these OEN numbers assigned for that S-EL1 at init time via some device tree or fixed structure approach?
Using above approach will avoid any contention among multiple trusted OS.

Please do let me know if I missed something in my understanding.

danh-arm, 2017-07-20:
> Shouldn't it be the responsibility of ATF to provide these OEN numbers assigned for that S-EL1 at init time via some device tree or fixed structure approach?

That would make sense in a system that supported multiple Trusted OS, but as previously stated, ARM TF doesn't support this.

b49020, 2017-07-21:
@danh-arm Thanks for your comments. May I know if you have any plans to support multiple trusted OS loading in ATF.

danh-arm, 2017-07-25:
As alluded to earlier, possibly in a future architecture version. So don't hold your breath!

closed 2017-08-14 by danh-arm