ARM-software/tf-issues #510: Question: about roadmap of TBB

nobuhiro76, 2017-08-09:
I am new to ATF. I have two questions about roadmap of TBB.

(1) Do you have plan to support RSA-3072 at the Trusted Board Boot sequence?
(2) Do you have plan to support AES decryption at boot sequence?

I would like to store encrypted BL32 image in flash memory and decrypt it after loading. 
Moreover, I would like to use AES instruction supported by ARMv8 Cryptography Extensions to shorten boot time. 

If you have above plans, please tell me the schedule.

danh-arm, 2017-08-14:
Hi @nobuhiro76 

In general we do not discuss ARM roadmap information in public. If you wish to understand or influence the Trusted Firmware roadmap, please contact your partner manager.

Having said that, I can answer your specific questions: We currently do not have any plans to implement firmware encryption or to use RSA-3072. If we did implement firmware encryption, this would use the mbed TLS APIs, therefore any AES hardware acceleration would be dependent on the mbed TLS implementation. Currently, mbed TLS does not implement AES hardware acceleration using the ARMv8-A crypto extensions.

Of course, there's nothing stopping you or others from implementing this; we're happy to advise where we can.

danh-arm, 2017-09-13:
I'm hoping this question is answered

closed 2017-09-13 by danh-arm