ARM-software/tf-issues #511: juno: Build error: BL2 image has exceeded its limit.

sorenb-xlnx, 2017-08-14:
I see juno builds failing when debug is enabled and a high log level selected. In that case the BL2 image exceeds the available memory:
```
aarch64-none-elf-ld: build/juno/debug/bl2/bl2.elf section `xlat_table' will not fit in region `RAM'
aarch64-none-elf-ld: BL2 image has exceeded its limit.
aarch64-none-elf-ld: region `RAM' overflowed by 4096 bytes
```

This can be reproduced with `make PLAT=juno SPD=opteed LOG_LEVEL=50 DEBUG=1`.

This happens since:
```
54661cd2483a62b8df7b11b6fedbdb5e29bbedbf is the first bad commit
commit 54661cd2483a62b8df7b11b6fedbdb5e29bbedbf
Author: Summer Qin <summer.qin@arm.com>
Date:   Mon Apr 24 16:49:28 2017 +0100

    Add Trusted OS extra image parsing support for ARM standard platforms

    Trusted OS may have extra images to be loaded. Load them one by one
    and do the parsing. In this patch, ARM TF need to load up to 3 images
    for optee os: header, pager and paged images. Header image is the info
    about optee os and images. Pager image include pager code and data.
    Paged image include the paging parts using virtual memory.

    Change-Id: Ia3bcfa6d8a3ed7850deb5729654daca7b00be394
    Signed-off-by: Summer Qin <summer.qin@arm.com>

:040000 040000 0b19cf90141beae09b1d9ff17f7ea49195165bf5 8594076510593dfaa5812be6f532b28809a950de M      include
:040000 040000 25155bf209399d245d8a570a6f1b9b688a1c6769 1c2c0ce2ceae315adb7602af92898ab8f2c57b27 M      lib
:040000 040000 146cdf05bb968d32a7e72b9427e022ebd107508c c7158c15131243b1d36e5ab4ae07f2cdb71263f2 M      plat
```

soby-mathew, 2017-08-16:
Yes, unforunately memory optimization related to assertion prints ([here](https://github.com/ARM-software/arm-trusted-firmware/blob/master/include/lib/stdlib/assert.h#L55)) is undone by selecting LOG_LEVEL=50 and Juno is tight on free SRAM memory. This can be fixed by defining PLAT_LOG_LEVEL_ASSERT to 40 in Juno platform_def.h and Juno must do this to save memory. Thanks for reporting this.

rangewang, 2017-08-21:
Hi sorenb-xlnx

Your error message is come from this [linker](https://github.com/ARM-software/arm-trusted-firmware/blob/master/bl2/bl2.ld.S#L151).

If your sram is bigger enough, maybe you can try to modify "BL2_LIMIT" value in [HERE](https://github.com/ARM-software/arm-trusted-firmware/blob/master/include/plat/arm/common/arm_def.h) or in your platform setting.

closed 2017-09-05 by davidcunado-arm