ARM-software/tf-issues #514: Function "pm_fpga_load" arguments different order

JoseRivas-R, 2017-08-29:
Hi,
It seems like on plat/xilinx/zynqmp/pm_service/pm_api_sys.c arguments of "pm_fpga_load" are in different order to function definition on pm_api_sys.h

On declaration: 'address_high, address_low, size, flags'
On definition: 'address_low, address_high, size, flags'

See images.

Thank you,
Regards
[image]
[image]

danh-arm, 2017-09-13:
Hi @sorenb-xlnx . Any comment on this?

sorenb-xlnx, 2017-09-18:
Yes, seems something went wrong somewhere. While I haven't personally tested this feature, AFAIK, it is working. Hence, the header should be changed to match the definition.
If nobody beats me to it, I'll create a PR.

closed 2017-09-19 by sorenb-xlnx