ARM-software/tf-issues #515: HiKey LeMaker with HiSi620 - GPIO interrupt support in secure world ?

ppeiffer, 2017-09-05:
Hi,

Is there a support for GPIO interrupt in secure world there ?
If yes, could you point me to some code sample ?

Thanks

danh-arm, 2017-09-13:
Hi

I don't know much about this platform or how compatible it is with the upstream HiKey code. However, I know that the upstream platform code does some GPIO initialisation. See `hikey_gpio_init()` in `plat/hisilicon/hikey/hikey_bl1_setup.c`.

@hzhuang1 - can you help with this question?

Dan.

hzhuang1, 2017-09-13:
Let's open the issue for a while. I've forward this issue to hisilicon.

I think this issue is focus on whether GPIO pin could be accessed in secure world. I don't know the details. And I hope that hisilicon guys could give me more information.

@danh-arm Yes, the GPIO driver is already enabled in HiKey platform. But I don't know whether all GPIO pins could be accessed in secure world or not. Peter could give a try first.

danh-arm, 2018-02-22:
Hi @hzhuang1. Did you here anything back about this? If not, I will close.

hzhuang1, 2018-02-23:
@ppeiffer It's not a good place to track the HiKey issue at here. Please help to create a bug in 96boards bug system. Then it is easy to track.
@danh-arm I can't bring anything back from Hisilicon. Please close it. We can track it in 96boards bug system.

closed 2018-02-26 by danh-arm