ARM-software/tf-issues #518: opteed assert failed

prime-zeng, 2017-09-20:
Hi:

I have met the following issue accidentally :
ASSERT: opteed_synchronous_sp_exit <120> : cm_get_context(SECURE) == &optee_ctx->cpu_ctx

I guess maybe is the cached issue,  maybe after the cpu enters the OP-TEE, it invalidate the cache and the data in ATF is lost, but i can't find the exact clue.
Any ideas?

Thank you.

soby-mathew, 2017-10-27:
@prime-zeng, as explained in https://github.com/OP-TEE/optee_os/issues/1824 , OP-TEE only invalidates its own data and so it shouldn't have affected data in ARM-TF.

You could try flushing the per-cpu data here :
https://github.com/ARM-software/arm-trusted-firmware/blob/f132b4a05b23916c1101add4bd6d973a99983719/services/spd/opteed/opteed_common.c#L37

After the cm_set_context(), add `flush_cpu_data(cpu_context[SECURE]);`  and see if the situation improves. If it indeed improves, then there something going with the caches which needs to be looked into.

danh-arm, 2018-02-22:
This issue has stalled. Please re-open if there's still a problem

closed 2018-02-22 by danh-arm