ARM-software/tf-issues #519: ATF does not build correctly with toolchains that have PIE enabled (currently Debian Stretch)

stevecapperlinaro, 2017-09-25:
Hello,
If one builds ATF using a version of gcc configured to use PIE by default, then the resultant binaries will not execute correctly (the system usually hangs).

One can determine whether or not gcc will use PIE from gcc -v:
```
$ aarch64-linux-gnu-gcc -v
Using built-in specs.
COLLECT_GCC=aarch64-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc-cross/aarch64-linux-gnu/6/lto-wrapper
Target: aarch64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 6.3.0-18'  ... --enable-default-pie ...
Thread model: posix
gcc version 6.3.0 20170516 (Debian 6.3.0-18)
```

(the presence of ```--enable-default-pie``` indicates PIE will be enabled by default)

One can fix ATF for use with such toolchains by adding ```-fno-PIE``` to ```TF_CFLAGS```.

Cheers,
Steve

davidcunado-arm, 2017-10-05:
Internal ref: https://jira.arm.com/browse/GENFW-2084

danh-arm, 2017-10-09:
This issue has come up before. The concern then was we can't apply "-fno-pie" unconditionally, since older version of GCC don't understand this flag (and have PIE disabled by default). As Steve suggests, we can probe whether PIE is enabled by default in the installed GCC version before each build, and use this to control TF_CFLAGS. It's a bit messy but should work.

Andre-ARM, 2017-11-17:
You might use some simple Makefile based toolchain feature detection like this: https://github.com/apritzel/arm-trusted-firmware/commit/3efb52570e581d79f6e451ef88933423a61a9b55
The kernel uses this heavily to work with a wide range of compilers.

closed 2017-12-19 by davidcunado-arm

afaerber, 2018-02-10:
This way of detecting does not work for openSUSE Tumbleweed, where `--enable-default-pie` is not present but instead a `defaults.spec` file gets installed via opt-out `gcc-PIE` package to enable it:
```
*default_spec:
%%{pie|fpic|fPIC|fpie|fPIE|fno-pic|fno-PIC|fno-pie|fno-PIE|shared|static|nostdlib|nodefaultlibs|nostartfiles:;:-fPIE -pie}
```
https://build.opensuse.org/package/view_file/openSUSE:Factory/gcc/gcc.spec?expand=1

So shouldn't we rather detect whether the compiler supports the `-fno-PIE` option and then always supply it?

danh-arm, 2018-02-14:
Hi @afaerber 

You may be right, but others may not see your comment as this issue is closed. I suggest you raise a new issue and/or submit a patch to address this.