ARM-software/tf-issues #522: Point release questions.

heeleyth, 2017-10-09:
Hi,

And visibility on when V1.5 is expected to be released?

Any plans for a regular point release cadence?

What if any is the extra validation that is performed on a point release?

Regards,

Tom.

danh-arm, 2017-10-09:
Hi Tom

>And visibility on when V1.5 is expected to be released?

We're not precisely sure yet but it's likely to be early next year.

> Any plans for a regular point release cadence?

It's something we're considering but no firm plans as yet. It requires a significant amount of additional effort, in particular around test coverage and automation. We've not had much feedback about it during roadmap discussions with partners.

> What if any is the extra validation that is performed on a point release?

We execute our full suite of automated tests, a significant number of manually executed tests, and an out-of-box test on the final release candidate.

Regards

Dan

heeleyth, 2017-10-09:
Hi Dan,

Thanks for the prompt response and I hope to see V1.5 early next year!

I understand the resourcing issue with regular cadence but it gives a great feeling of confidence (once a quarter would be nice!).

Closing this ticket (issue as a term can be negative!)

Tom.

closed 2017-10-09 by heeleyth