ARM-software/tf-issues #523: separating bl32.bin and/or bl33.bin from fip.bin

vchong, 2017-10-12:
Can bl32.bin and/or bl33.bin be stored and loaded separately from fip.bin?

danh-arm, 2017-10-12:
If you mean can these images be extracted from fip.bin, then "fiptool unpack" is your friend.

vchong, 2017-10-12:
No. I mean store each of bl32.bin, bl33.bin and fip.bin (basically bl2.bin only at this point) as 3 separate images in memory, and then load them from bl2.

danh-arm, 2017-10-12:
OK, I see what mean. To do this you override `plat_get_image_source()` to return an IO device handle and an `io_block_spec_t` corresponding to the provided image id (defined in `tbbr_img_def.h`). The `io_block_spec_t` for each image will need to specify the address and size in memory.

vchong, 2017-10-12:
Thanks! Will give it a try.

closed 2018-02-22 by danh-arm