ARM-software/tf-issues #525: Fix `warning: 'plat_get_syscnt_freq' is deprecated`

paulmenzel, 2017-10-12:
The deprecation warning below is shown.

```
$ git log --oneline -1 origin/master
9679297f Merge pull request #1120 from michpappas/tf-issues#521_cert_tool_does_not_build_with_openssl_v1.1
$make PLAT=mt8173
/usr/bin/aarch64-linux-gnu-gcc -DDEBUG=0 -DNDEBUG -DSTACK_PROTECTOR_ENABLED=0 -DARM_GIC_ARCH=2 -DMTK_SIP_SET_AUTHORIZED_SECURE_REG_ENABLE -DSKIP_A57_L1_FLUSH_PWR_DWN=0 -DA53_DISABLE_NON_TEMPORAL_HINT=1 -DA57_DISABLE_NON_TEMPORAL_HINT=1 -DERRATA_A53_826319=1 -DERRATA_A53_835769=0 -DERRATA_A53_836870=1 -DERRATA_A53_843419=0 -DERRATA_A53_855873=1 -DERRATA_A57_806969=0 -DERRATA_A57_813419=0 -DERRATA_A57_813420=0 -DERRATA_A57_826974=0 -DERRATA_A57_826977=0 -DERRATA_A57_828024=0 -DERRATA_A57_829520=0 -DERRATA_A57_833471=0 -DERRATA_A57_859972=0 -DERRATA_A72_859971=0 -DCRASH_REPORTING=0 -DARM_ARCH_MAJOR=8 -DARM_ARCH_MINOR=0 -DARM_GIC_ARCH=2 -DCOLD_BOOT_SINGLE_CPU=0 -DCTX_INCLUDE_AARCH32_REGS=1 -DCTX_INCLUDE_FPREGS=0 -DENABLE_ASSERTIONS=0 -DENABLE_PLAT_COMPAT=1 -DENABLE_PMF=0 -DENABLE_PSCI_STAT=0 -DENABLE_RUNTIME_INSTRUMENTATION=0 -DENABLE_SPE_FOR_LOWER_ELS=0 -DERROR_DEPRECATED=0 -DHW_ASSISTED_COHERENCY=0 -DLOAD_IMAGE_V2=0 -DLOG_LEVEL=20 -DNS_TIMER_SWITCH=0 -DPL011_GENERIC_UART=0 -DPLAT_mt8173 -DPROGRAMMABLE_RESET_ADDRESS=1 -DPSCI_EXTENDED_STATE_ID=0 -DRESET_TO_BL31=0 -DSEPARATE_CODE_AND_RODATA=0 -DSPD_none -DSPIN_ON_BL1_EXIT=0 -DTRUSTED_BOARD_BOOT=0 -DUSE_COHERENT_MEM=1 -DUSE_TBBR_DEFS=0 -DWARMBOOT_ENABLE_DCACHE_EARLY=0 -DAARCH64 -Iinclude/lib/stdlib -Iinclude/lib/stdlib/sys -Iinclude/bl1 -Iinclude/bl31 -Iinclude/common -Iinclude/common/aarch64 -Iinclude/drivers -Iinclude/drivers/arm -Iinclude/drivers/auth -Iinclude/drivers/io -Iinclude/drivers/ti/uart -Iinclude/lib -Iinclude/lib/aarch64 -Iinclude/lib/cpus -Iinclude/lib/cpus/aarch64 -Iinclude/lib/el3_runtime -Iinclude/lib/el3_runtime/aarch64 -Iinclude/lib/pmf -Iinclude/lib/psci -Iinclude/lib/xlat_tables -Iinclude/plat/common -Iinclude/services -Iplat/mediatek/common/ -Iplat/mediatek/common/drivers/uart/ -Iinclude/plat/arm/common -Iinclude/plat/arm/common/aarch64 -Iplat/mediatek/mt8173/drivers/crypt/ -Iplat/mediatek/mt8173/drivers/mtcmos/ -Iplat/mediatek/mt8173/drivers/pmic/ -Iplat/mediatek/mt8173/drivers/rtc/ -Iplat/mediatek/mt8173/drivers/spm/ -Iplat/mediatek/mt8173/drivers/timer/ -Iplat/mediatek/mt8173/include/  -Iinclude/tools_share -nostdinc -Wmissing-include-dirs -Werror -march=armv8-a -mgeneral-regs-only -ffreestanding -fno-builtin -Wall -std=gnu99 -Os -ffunction-sections -fdata-sections -Wno-error=deprecated-declarations  -DIMAGE_BL31 -Wp,-MD,./build/mt8173/release/bl31/plat_common.d -MT build/mt8173/release/bl31/plat_common.o -MP -c plat/common/aarch64/plat_common.c -o build/mt8173/release/bl31/plat_common.o
plat/common/aarch64/plat_common.c: In function 'plat_get_syscnt_freq2':
plat/common/aarch64/plat_common.c:60:2: warning: 'plat_get_syscnt_freq' is deprecated [-Wdeprecated-declarations]
  unsigned long long freq = plat_get_syscnt_freq();
  ^~~~~~~~
In file included from plat/common/aarch64/plat_common.c:8:0:
include/plat/common/platform.h:41:20: note: declared here
 unsigned long long plat_get_syscnt_freq(void) __deprecated;
                    ^~~~~~~~~~~~~~~~~~~~
```

It’s very strange though, as the used method doesn’t seem to be implemented in ARM Trusted Firmware.

```
$ git grep 'plat_get_syscnt_freq('
include/plat/common/platform.h:unsigned long long plat_get_syscnt_freq(void) __deprecated;
plat/common/aarch64/plat_common.c:      unsigned long long freq = plat_get_syscnt_freq();
```

davidcunado-arm, 2017-11-20:
Hi @paulmenzel 
Apologies for slow reply... What's happening is as follows when ERROR_DEPRECATED = 0:
- plat_common.c is providing a weak implementation of plat_get_syscnt_freq2 that calls plat_get_syscnt_freq (which is deprecated).
- For ARM platforms (and in fact for all the platform ports in /plat), there is a strong implementation of plat_get_syscnt_freq2 - therefore the linker ignores the weak implementation and the missing implementation of plat_get_syscnt_freq is not an issue. (NOTE - for ARM platforms, this string implementation is in arm_common.c).
- A platform that does not provide a strong implementation of plat_get_syscnt_freq2 is expected to provide an implementation of plat_get_syscnt_freq.

With ERROR_DEPRCATED = 1:
- The implementation of plat_get_syscnt_freq2 in plat_common.c is not compiled in.
- The implementation of plat_get_syscnt_freq2 in arm_common.c becomes the weak implementation for ARM platforms.

I hope this helps!

David

Andre-ARM, 2017-11-20:
Chiming in here because I was stumbling upon the same problem.

>What's happening is as follows when ERROR_DEPRECATED = 0:
> - plat_common.c is providing a weak implementation of plat_get_syscnt_freq2 that calls
> plat_get_syscnt_freq (which is deprecated).

Yes, but there is not a single implementation of the old plat_get_syscnt_freq() in the upstream tree.

> - For ARM platforms (and in fact for all the platform ports in /plat), there is a strong implementation
> of plat_get_syscnt_freq2 - therefore the linker ignores the weak implementation and the [...]

I agree to that idea, only that it is not working as intended. I guess it's due to the *linker* having to throw away the weak implementation, but the *compiler* still generating the weak version, tripping over the __deprecated plat_get_syscnt_freq() header and generating the warning.

`$ make PLAT=juno ERROR_DEPRECATED=0 all`

shows that.
 
> With ERROR_DEPRCATED = 1:
> 
> - The implementation of plat_get_syscnt_freq2 in plat_common.c is not compiled in.
> - The implementation of plat_get_syscnt_freq2 in arm_common.c becomes the weak implementation for ARM platforms.

So while this is true (and indeed adding ERROR_DEPRECATED=1 to the make command line or platform.mk avoids the warning) I was wondering if this is the right solution. My (naive?) understanding of ERROR_DEPRECATED is that it points out the usage of deprecated functions by *breaking* compilation. Leaving it at 0 should still point out the usage of deprecated functions, though not be fatal (possibly only warn). **But**: there is no deprecation violation for Juno (and all the other platforms) - short of that legacy stub. So I was wondering if that weak function should be protected by  ENABLE_PLAT_COMPAT instead, though I am not sure this the correct scope of this symbol.

danh-arm, 2017-11-21:
The implementation is working as intended but yes the warnings are a bit confusing. In general, most warnings are treated as errors in the build (-Werror) but these specific warnings are ignored by the build system (-Wno-error=deprecated-declarations). They are a side effect of providing backwards compatibility.

In a future incompatible release, we'd like to remove all deprecated code under the ERROR_DEPRECATED flag (and ENABLE_PLAT_COMPAT), including this weak implementation of plat_get_syscnt_freq2(). However, I'd like this all to be done at the same time unless there's a specific reason to do this piecemeal. Doing this just to make the build log look cleaner is not a good reason. However, perhaps it's possible to supress the warnings in these specific cases where we know there's a good reason to use the deprecated functions. A contribution to that effect would be welcome.

Also note that this project provides stronger API compatibility guarantees than most other projects since we expect most platform code to always be out of tree and a common use-model for some platforms is to stay on the same version of generic code for a long time. If we only cared about upstream platforms we could be more aggressive in removing deprecated code.

Andre-ARM, 2017-11-21:
@danh-arm, thanks for the clarification! I can understand the deprecation policy and am totally fine with it.
However I would strongly advise against ignoring warnings. Starting to be sloppy here really asks for trouble: genuine warnings are then missed, because they are buried in "OK to ignore" warnings (been there, costed me a week once). And it becomes harder to argue against contributions where "compilation succeeds with warnings" if you give bad examples.

So I wanted to look for a clever fix, but was wondering if you could confirm what the intended meaning of ERROR_DEPRECATED is, exactly?
1) Upstream platforms without legacy code generate no warnings at all, regardless of the ERROR_DEPRECATED setting. This is because they are doing nothing wrong.
2) Downstream platforms with a legacy plat_get_syscnt_freq() implementation:
2a) generate a warning on compilation with ERROR_DEPRECATED=0.
2b) fail to compile with ERROR_DEPRECATED=1.

Is that about right? Or should 2a) be completely silent (no warnings)?

And in general: Are there other out-of-tree users of functions like  plat_get_syscnt_freq() we need to consider? Like secure EL1 payloads? Which downstream users keep in *their* tree? Or is this purely about the plat/ directory?

danh-arm, 2017-11-21:
We don't ignore warning generally. We specifically only ignore the deprecated warnings when ERROR_DEPRECATED=0 to give a heads up that platforms should "do something". Even these warnings generate an error when ERROR_DEPRECATED=1. All other warnings generate errors.

What you propose in 2 would be nice, but I think it's going to be hard to get downstream platforms to generate a warning/error if they define a plat_get_syscnt_freq() (or other deprecated function). The deprecated warnings are only generated when deprecated functions are called, not defined.

I think to achieve 1, it woud be possible to suppres the warning in the plat/common implementation of  plat_get_syscnt_freq2() by using something like this: 

#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wdeprecated-declarations"
	unsigned long long freq = plat_get_syscnt_freq();
#pragma GCC diagnostic pop

We already do someting similar in psci_set_req_local_pwr_state() in psci_common.c.

danh-arm, 2017-11-21:
> #pragma GCC diagnostic push
> #pragma GCC diagnostic ignored "-Wdeprecated-declarations"
> unsigned long long freq = plat_get_syscnt_freq();
> #pragma GCC diagnostic pop

I just tried this and it works!

Andre-ARM, 2017-11-21:
> We don't ignore warning generally. We specifically only ignore the deprecated warnings when
> ERROR_DEPRECATED=0 to give a heads up that platforms should "do something".

Well, Juno and most other platforms are fine, they don't need to do something. Still they generate warnings. I don't think that's necessary.

> I think to achieve 1, it woud be possible to suppres the warning in the plat/common implementation of plat_get_syscnt_freq2() by using something like this:
>
> #pragma GCC diagnostic push
> #pragma GCC [...]

Yeah, I saw this, but to be honest I don't like this too much. It's a bit weird that we first ask the compiler to warn us about the usage of deprecated functions, then tell it to suppress the warning.
I believe there are better ways, I have some ideas in mind...

antonio-nino-diaz-arm, 2018-09-28:
This problem will be fixed in https://github.com/ARM-software/arm-trusted-firmware/pull/1587

I'm closing this issue now in case I forget.

closed 2018-09-28 by antonio-nino-diaz-arm