ARM-software/tf-issues #529: Uniphier: Possible dependency issue in the Makefile

sandrine-bailleux-arm, 2017-11-03:
Hi @masahir0y

One of our CI jobs failed this morning while attempting to build the Uniphier platform port. The error message was the following:

```
OPENSSL build/uniphier/debug/rot_key.pem
/bin/sh: 1: cannot create build/uniphier/debug/rot_key.pem: Directory nonexistent
```

The command line used to build TF was:

```
make -j DEBUG=1 LOG_LEVEL=50 TRUSTED_BOARD_BOOT=1 GENERATE_COT=1 MBEDTLS_DIR=mbedtls SPD=tspd all
```

Unfortunately, I couldn't reproduce the issue locally on my machine and rerunning the CI job passed the second time. I suspect the problem might be due to a missing dependency in one of the the Uniphier makefile rules. As far as I understand, the build system attempted to generated the Root of Trust key `($(BUILD_PLAT)/rot_key.pem)` while the build directory (`$(BUILD_PLAT)`) didn't exist yet. I think the issue might be exacerbated when doing a parallel build (using the -j option of make).

Regards,
Sandrine

masahir0y, 2017-11-03:
@sandrine-bailleux-arm 

Thanks for the report!
I pushed PR #1155 to fix this problem.

antonio-nino-diaz-arm, 2017-11-22:
The PR has been merged. https://github.com/ARM-software/arm-trusted-firmware/pull/1155

closed 2017-11-22 by antonio-nino-diaz-arm