ARM-software/tf-issues #530: PSCI call CPU_OFF sometimes doesn't perform a clean shutdown.

JorgeMVP, 2017-11-09:
I have noticed that when a CPU power itself down via CPU_OFF, the ATF (PSCI implementation 0.2) continues returning the state ON in response to the call AFFINITY_INFO from the other CPUs.

I was expecting any return in case of a unsuccessfully call CPU_OFF, however, it doesn't return and from the point of view of PSCI implementation the state is always ON.

I don't know if it was solved in the new versions (1.0/1.1) or is it only the (PSCI implementation 0.2) which has this issue?

robertovargas-arm, 2017-11-13:
It is hard for us to fix the problem that you comment because the code and data structures are very different in the current version that supports PSCI 1.1. Although we cannot reproduce the problem with our platforms, we had identified a theoretical problem with cache coherency of the PSCI data, which appears similar to yours. You can take a look to [1] because the problem that this patch solved for CPU_ON can happen in AFFINITY_INFO.

[1] https://github.com/ARM-software/arm-trusted-firmware/commit/71341d23668b1700cd50c0abd74073e4351acbd9

JorgeMVP, 2017-11-14:
Thank you so much, I will try your tip and then I will be back to give feedback.

danh-arm, 2018-02-22:
We're didn't hear anything back about this. Please reopen if you still have an issue

closed 2018-02-22 by danh-arm