ARM-software/tf-issues #531: PSCI implementation 1.1 of SYSTEM_OFF

JorgeMVP, 2017-11-15:
Hi all,

I have some doubts about the PSCI implementation of SYSTEM_OFF since it isn't that clear for me. 

According with PSCI 1.1 spec. for SYSTEM_OFF requires the caller to ensure that the cores are in a known state. 

Then it gives an example: 

> One way in which cores can be placed into a known state is to use calls to CPU_OFF on all online cores except for the last one, which instead uses SYSTEM_OFF. 

In this case, all cores will be in OFF state. My question is what is a known state for you? Are ON and ON_PENDING a known state as well?

In case of the caller doesn't place the remaining cores in a known state, what should happen? Should the PSCI implementation fail in response to PSCI call SYSTEM_OFF or force the system shutdown anyway?

Thanks in advance,
Jorge

robertovargas-arm, 2017-11-15:
It is not a requirement, it is an advice to the caller (usually the operating
system). If it doesn't set the cores in a known state (whatever it is for the
kernel, although the most common state should be CPU_OFF) it is going to have
problems once the system resumes (filesystem fails, race conditions, ...).
Once the caller calls SYSTEM_OFF the system *must* suspend, independently of
the state of the cores.

Kind regards,
Roberto Vargas.

JorgeMVP, 2017-11-15:
Oh ok, thank you so much.

Now everything makes sense.

Thanks once again.
Jorge

closed 2017-11-20 by robertovargas-arm