ARM-software/tf-issues #533: HiKey LeMaker Recovery Failed

XevenQC, 2017-12-02:
I am trying enabling OPTEE on hikey LeMaker verison. After I refalshed fip.bin, the device got stuck at "BL3-1: Initializing BL3-2". But when I trying to flash back with the recovery mode, it cannot got into the this mode. And I tried all combinations of jumpers, it cannot reflash this device any more. Accoding the source of arm-trust-firmware, fastboot protocol is implemented in bl1. How can we get into this mode when the fip in damged and failed to boot to bl33. I tried several devices and got the same issue. It seems that there is hardware bug for hikey LeMaker version.

Could you show me how to reflash the device if the fip.bin is damaged? Thanks!

hzhuang1, 2017-12-04:
Please make sure that you're following the document (https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/plat/hikey.rst).

XevenQC, 2017-12-04:
Yes, I have followed this doc. It still cannot got into fastboot mode and fastboot flash command is always waiting for device.

hzhuang1, 2017-12-04:
Press ESCAPE for boot options .Android Fastboot mode - version 0.6.
Press RETURN or SPACE key to quit.
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v1.4(release):v1.4-410-g5627c1e
NOTICE:  BL1: Built : 21:43:05, Dec  4 2017
NOTICE:  BL1-FWU: *******FWU Process Started*******
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v1.1(release):5b0d44c
NOTICE:  BL1: Built : 21:41:56, Dec  4 2017
NOTICE:  syspll frequency:1190494208Hz
NOTICE:  succeed to init lpddr3 rank0 dram phy
INFO:    lpddr3_freq_init, set ddrc 533mhz
INFO:    init ddr3 rank0
INFO:    ddr3 rank1 init pass
INFO:    lpddr3_freq_init, set ddrc 800mhz
INFO:    init ddr3 rank0
INFO:    ddr3 rank1 init pass
INFO:    Samsung DDR
NOTICE:  Enter fastboot mode...

It works well with latest commit in ATF. I doubt that you missed something.

SevenQC, 2017-12-04:
Thanks for your kindly reply. My problem is that when command "sudo python hisi_idt.py -d /dev/ttyUSB0 --img1=./l-loader.bin" finished and the images has been loaded into RAM. But l-loader.bin does not got executed. There is nothing in console log. And I have linked both pin 1-2 and 3-4

hzhuang1, 2017-12-04:
Oh, I see. Check your cables first. 
* If two USB cables are connected to your hikey. ttyUSB0 is your serial console, and ttyUSB1 is your OTG cable.
* If only one USB cable is connected to your hikey. ttyUSB0 is your OTG cable.
It seems that you're using ttyUSB0 as OTG. And you still expects console output. It means that you're connecting two cables to hikey.

SevenQC, 2017-12-04:
No, I connected two USB cables to hikey. One is for USB OTG and anther is for UART. Only UART can be recognized as ttyUSB0 and USB OTG cannot be recognized.

vchong, 2017-12-04:
Are you using virtual linux machine?

SevenQC, 2017-12-04:
No, Ubuntu 16.04 is installed on my PC. So is there something wrong with my PC? But before reflashing the images, it can be recognized by fastboot over USB OTG.

vchong, 2017-12-04:
Try recovering again with `http://releases.linaro.org/reference-platform/embedded/hikey/latest/rpb/bootloader/l-loader.bin`. Maybe you have a bad/corrupted l-loader.bin.

XevenQC, 2017-12-05:
Hi vchong, thanks for your help.

I tried to recovery with the l-loader.bin provided by you. It makes no difference. Is it ok that USB OTG cannot be recognized in recovery mode (pin 1-2 and 3-4 are closed) ?  It seems something wrong with my PC, the USB OTG can be recognized for a very short while, as the following dmesg log.

[ 2734.123494] usb 1-1: new full-speed USB device number 34 using xhci_hcd
[ 2734.141135] usb 1-1: New USB device found, idVendor=12d1, idProduct=3609
[ 2734.141147] usb 1-1: New USB device strings: Mfr=1, Product=4, SerialNumber=0
[ 2734.141154] usb 1-1: Product: \xffffffe3\xffffff84\xffffffb0㌲㔴㜶㤸
[ 2734.141158] usb 1-1: Manufacturer: 䕇䕎䥎
[ 2734.142091] option 1-1:1.0: GSM modem (1-port) converter detected
[ 2734.142797] usb 1-1: GSM modem (1-port) converter now attached to ttyUSB0
[ 2734.711244] usb 1-1: USB disconnect, device number 34
[ 2734.711413] option1 ttyUSB0: GSM modem (1-port) converter now disconnected from ttyUSB0
[ 2734.711431] option 1-1:1.0: device disconnected

hzhuang1, 2017-12-05:
> [ 2734.123494] usb 1-1: new full-speed USB device number 34 using xhci_hcd
> [ 2734.141135] usb 1-1: New USB device found, idVendor=12d1, idProduct=3609
> [ 2734.141147] usb 1-1: New USB device strings: [...]

As I said, OTG should connect to ttyUSB1. But it's _JUST_ ttyUSB0 on your board. Please make sure connect serial port cable first, then attach USB OTG cable.

XevenQC, 2017-12-05:
Hi zhuang, as you said I attached two usb cables, first UART, and the USB OTG. And the USB OTG  still can be recognized for a very short while, as the following dmesg log.

[ 5259.599613] usb 1-3.4: new full-speed USB device number 42 using xhci_hcd
[ 5259.624035] usb 1-3.4: New USB device found, idVendor=0403, idProduct=6015
[ 5259.624038] usb 1-3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5259.624040] usb 1-3.4: Product: FT230X 96Boards Console
[ 5259.624041] usb 1-3.4: Manufacturer: FTDI
[ 5259.624042] usb 1-3.4: SerialNumber: DAVT3ZB
[ 5259.628440] ftdi_sio 1-3.4:1.0: FTDI USB Serial Device converter detected
[ 5259.628461] usb 1-3.4: Detected FT-X
[ 5259.628463] usb 1-3.4: Number of endpoints 2
[ 5259.628464] usb 1-3.4: Endpoint 1 MaxPacketSize 64
[ 5259.628465] usb 1-3.4: Endpoint 2 MaxPacketSize 64
[ 5259.628466] usb 1-3.4: Setting MaxPacketSize 64
[ 5259.628657] usb 1-3.4: FTDI USB Serial Device converter now attached to ttyUSB0
[ 5276.334228] usb 1-3.2: new full-speed USB device number 43 using xhci_hcd
[ 5276.351935] usb 1-3.2: New USB device found, idVendor=12d1, idProduct=3609
[ 5276.351941] usb 1-3.2: New USB device strings: Mfr=1, Product=4, SerialNumber=0
[ 5276.351944] usb 1-3.2: Product: \xffffffe3\xffffff84\xffffffb0㌲㔴㜶㤸
[ 5276.351946] usb 1-3.2: Manufacturer: 䕇䕎䥎
[ 5276.353280] option 1-3.2:1.0: GSM modem (1-port) converter detected
[ 5276.353396] usb 1-3.2: GSM modem (1-port) converter now attached to ttyUSB1
[ 5277.158836] usb 1-3.2: USB disconnect, device number 43
[ 5277.158974] option1 ttyUSB1: GSM modem (1-port) converter now disconnected from ttyUSB1
[ 5277.159006] option 1-3.2:1.0: device disconnected

And I repeated plug/unplug the OTG cables for several times, it finally works. Maybe there is something wrong with my PC. Greate thanks for your guys kindly help.

closed 2017-12-05 by XevenQC