ARM-software/tf-issues #535: Boot crashes on Hikey

soby-mathew, 2017-12-07:
@Leo-Yan, @hzhuang1 

I am following the instructions mentioned to build and boot Hikey960 : https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/plat/hikey960.rst

But when I boot UEFI, I get a synchronous exception. I also tried to flash pre-built binaries from https://builds.96boards.org/snapshots/reference-platform/components/uefi-staging/26/hikey960/debug/ but still have the same crash.

Any clues what is wrong ?

```
..........[Bds]Booting Boot from SD
[Bds]Booting Grub
[Bds]Booting Android Boot
add-symbol-file /home/buildslave/workspace/96boards-reference-uefi-staging/26/edk2/Build/HiKey960/DEBUG_GCC49/AARCH64/EmbeddedPkg/Application/AndroidBoot/AndroidBootApp/DEBUG/AndroidBootApp.dll 0xBA175000
Loading driver at 0x000BA174000 EntryPoint=0x000BA175000 AndroidBootApp.efi

Synchronous Exception at 0x00000000BF0D783C
PC 0x0000BF0D783C (0x0000BF0CB000+0x0000C83C) [ 0] HiKey960Dxe.dll
PC 0x0000BF0CD120 (0x0000BF0CB000+0x00002120) [ 0] HiKey960Dxe.dll
PC 0x0000BA176744 (0x0000BA174000+0x00002744) [ 1] AndroidBootApp.dll
PC 0x0000BA1768D8 (0x0000BA174000+0x000028D8) [ 1] AndroidBootApp.dll
PC 0x0000BA175814 (0x0000BA174000+0x00001814) [ 1] AndroidBootApp.dll
PC 0x0000BA1753E0 (0x0000BA174000+0x000013E0) [ 1] AndroidBootApp.dll
PC 0x0000BA175064 (0x0000BA174000+0x00001064) [ 1] AndroidBootApp.dll
PC 0x0000BF1DD9EC (0x0000BF1D7000+0x000069EC) [ 2] DxeCore.dll
PC 0x0000BF02C344 (0x0000BF01B000+0x00011344) [ 3] BdsDxe.dll
PC 0x0000BF01CE90 (0x0000BF01B000+0x00001E90) [ 3] BdsDxe.dll
PC 0x0000BF01E564 (0x0000BF01B000+0x00003564) [ 3] BdsDxe.dll
PC 0x0000BF1D9358 (0x0000BF1D7000+0x00002358) [ 4] DxeCore.dll
PC 0x0000BF1D841C (0x0000BF1D7000+0x0000141C) [ 4] DxeCore.dll
PC 0x0000BF1D8024 (0x0000BF1D7000+0x00001024) [ 4] DxeCore.dll

[ 0] /home/buildslave/workspace/96boards-reference-uefi-staging/26/edk2/Build/HiKey960/DEBUG_GCC49/AARCH64/OpenPlatformPkg/Platforms/Hisilicon/HiKey960/HiKey960Dxe/HiKey960Dxe/DEBUG/HiKey960Dxe.dll
[ 1] /home/buildslave/workspace/96boards-reference-uefi-staging/26/edk2/Build/HiKey960/DEBUG_GCC49/AARCH64/EmbeddedPkg/Application/AndroidBoot/AndroidBootApp/DEBUG/AndroidBootApp.dll
[ 2] /home/buildslave/workspace/96boards-reference-uefi-staging/26/edk2/Build/HiKey960/DEBUG_GCC49/AARCH64/MdeModulePkg/Core/Dxe/DxeMain/DEBUG/DxeCore.dll
[ 3] /home/buildslave/workspace/96boards-reference-uefi-staging/26/edk2/Build/HiKey960/DEBUG_GCC49/AARCH64/MdeModulePkg/Universal/BdsDxe/BdsDxe/DEBUG/BdsDxe.dll
[ 4] /home/buildslave/workspace/96boards-reference-uefi-staging/26/edk2/Build/HiKey960/DEBUG_GCC49/AARCH64/MdeModulePkg/Core/Dxe/DxeMain/DEBUG/DxeCore.dll

  X0 0x00000001A67E00CE   X1 0x00000000BFFFF648   X2 0x00000000BF0CD104   X3 0x0000000000001000
  X4 0x00000000BE371000   X5 0x0000000000000000   X6 0x00000000BF0D1430   X7 0x00000000BFFFF4F4
  X8 0x00000000BE855BA8   X9 0x00000000BE6B3C20  X10 0x0000000000000004  X11 0x0000000000000000
 X12 0x00000000703FE07A  X13 0x0000000000000000  X14 0x0000000000000000  X15 0x0000000000000000
 X16 0x00000000BFFFF9D0  X17 0x0000000000000000  X18 0x0000000000000000  X19 0x00000000BE73B718
 X20 0x00000000BE73C218  X21 0x0000000000000000  X22 0x0000000000000000  X23 0x0000000000000000
 X24 0x0000000000000000  X25 0x0000000000000000  X26 0x0000000000000000  X27 0x0000000000000000
 X28 0x0000000000000000   FP 0x00000000BFFFF590   LR 0x00000000BF0CD120

  V0 0xE9B5DBA5B5C0FBCF 71374491428A2F98   V1 0xAB1C5ED5923F82A4 59F111F13956C25B
  V2 0x550C7DC3243185BE 12835B01D807AA98   V3 0xC19BF1749BDC06A7 80DEB1FE72BE5D74
  V4 0x240CA1CC0FC19DC6 EFBE4786E49B69C1   V5 0x76F988DA5CB0A9DC 4A7484AA2DE92C6F
  V6 0xBF597FC7B00327C8 A831C66D983E5152   V7 0x1429296706CA6351 D5A79147C6E00BF3
  V8 0x0000000000000000 2E1B213827B70A85   V9 0x0000000000000000 766A0ABB650A7354
 V10 0x0000000000000000 A81A664BA2BFE8A1  V11 0x0000000000000000 D6990624D192E819
 V12 0x0000000000000000 1E376C0819A4C116  V13 0x0000000000000000 4ED8AA4A391C0CB3
 V14 0x0000000000000000 78A5636F748F82EE  V15 0x0000000000000000 A4506CEB90BEFFFA
 V16 0x3C83709547545153 BC990E9F897D4FA8  V17 0xBEBEF297C5EC0578 E1D99AE8C2B92607
 V18 0x13242833C5F05BAB 101C24C521387481  V19 0x0A50ABCAD0BDDA12 996865483E880969
 V20 0x6C4ABAA53A9BE1CB 4416D9F479B08221  V21 0x60952147C3A68574 55B8AE51435C1A1A
 V22 0x9FEB2A3B4AB8D3BF 88C1883495C7F76F  V23 0xD0C224BC8FB77E09 3DB8D233CF470963
 V24 0x0E72963E02D21C93 C95B757DA62B3A12  V25 0x2A77DBA1E4EE5D5C E08479A1B557DFA8
 V26 0xB593F86668CA8129 927A0019CDEC1A76  V27 0xDAFF3CFC3673F75F 471FAB1A2F8E7BBF
 V28 0xE4EB07B5C6F71976 EE7A86EF3FFFFFDF  V29 0x453E29DFAF99657C C3E6AAFFBD7F0BEB
 V30 0xD7AB62CCD95F7DEF 8CD75D29FE27FFFF  V31 0xE6BDBCF3FF6ADDFF EEEBFF2CBFE63BCF

  SP 0x00000000BFFFF590  ELR 0x00000000BF0D783C  SPSR 0x60000209  FPSR 0x00000000
 ESR 0x96000005          FAR 0x00000001A67E00CE

 ESR : EC 0x25  IL 0x1  ISS 0x00000005

Data abort: Translation fault, first level

Stack dump:
  00000BFFFF490: 996865483E880969 0A50ABCAD0BDDA12 4416D9F479B08221 6C4ABAA53A9BE1CB
  00000BFFFF4B0: 55B8AE51435C1A1A 60952147C3A68574 88C1883495C7F76F 9FEB2A3B4AB8D3BF
  00000BFFFF4D0: 3DB8D233CF470963 D0C224BC8FB77E09 C95B757DA62B3A12 0E72963E02D21C93
  00000BFFFF4F0: E08479A1B557DFA8 2A77DBA1E4EE5D5C 927A0019CDEC1A76 B593F86668CA8129
  00000BFFFF510: 471FAB1A2F8E7BBF DAFF3CFC3673F75F EE7A86EF3FFFFFDF E4EB07B5C6F71976
  00000BFFFF530: C3E6AAFFBD7F0BEB 453E29DFAF99657C 8CD75D29FE27FFFF D7AB62CCD95F7DEF
  00000BFFFF550: EEEBFF2CBFE63BCF E6BDBCF3FF6ADDFF 00000000BF0D783C 0000000060000209
  00000BFFFF570: 0000000000000000 0000000096000005 00000001A67E00CE 00000000BF0CD0DC
> 00000BFFFF590: 00000000BFFFF5B0 00000000BF0CD120 00000000B9BFA620 00000001A67E00CE
  00000BFFFF5B0: 00000000BFFFF610 00000000BA176744 00000000BE73B718 00000000BE73C218
  00000BFFFF5D0: 00000000BFFFF648 00000001A67E00CE 00000000BE855B18 00000000BE85F53B
  00000BFFFF5F0: 00000000BE8490A8 00000000BE371000 00000000BE371000 0000000000000000
  00000BFFFF610: 00000000BFFFF870 00000000BA1768D8 0000000000000020 00000000BE370698
  00000BFFFF630: 00000001A67E00CE 00000000BD9BE000 59F111F13956C25B AB1C5ED5923F82A4
  00000BFFFF650: 6264696F72646E61 647261682E746F6F 6B69683D65726177 6F63203036397965
  00000BFFFF670: 74743D656C6F736E 6E61203051494679 6F6F6264696F7264 6C6F736E6F632E74
ASSERT [ArmCpuDxe] /home/buildslave/workspace/96boards-reference-uefi-staging/26/edk2/ArmPkg/Library/DefaultExceptionHandlerLib/AArch64/DefaultExceptionHandler.c(265): ((BOOLEAN)(0==1))
```

hzhuang1, 2017-12-07:
It tries to load kernel from boot partition. But you don't have the right kernel. Do you want to use Linux or AOSP?

soby-mathew, 2017-12-08:
I flashed the pre-built kernel fetched from this URL : http://builds.96boards.org/snapshots/hikey960/linaro/aosp-master/latest/hikey960-img-336.zip

I would like to use both Linux and AOSP eventually, but for now AOSP should suffice. Are there some pre-built kernel images for UEFI available ?

hzhuang1, 2017-12-08:
Kernel is in this zip file (http://builds.96boards.org/snapshots/hikey960/linaro/aosp-master/latest/hikey960-linaro-2017.12.03-factory-0346d8e8.zip).

If you want to run Linux, you could get it from this link (https://snapshots.linaro.org/reference-platform/embedded/morty/hikey960/latest/).

Prebuilt UEFI is in this link (http://builds.96boards.org/snapshots/reference-platform/components/uefi-staging/latest/hikey960/).

soby-mathew, 2017-12-08:
Thanks @hzhuang1. I have AOSP kernel booting now.

closed 2017-12-08 by soby-mathew