ARM-software/tf-issues #539: Shall we save scr_el3 in handle_interrupt_exception macro?

iwishguo, 2017-12-23:
Hi folks,

We don't see scr_el3 is saved in handle_interrupt_exception macro, but we can see it being restored from context in el3_exit. Is it a code bug?

Thanks,
Gary

davidcunado-arm, 2017-12-24:
Hi @iwishguo 

The values for SCR_EL3 are initialised in el3_arch_init_common (for the common values) and then the values updated for the next EL in cm_init_context_common, stored in the appropriate context object (for the Non-secure or Secure world) - this value is then restored on world switch in el3_exit, restored from the appropriate context.

Hope this helps!

David

iwishguo, 2017-12-26:
Hi @davidcunado-arm 

Thanks for your explanation.
Do you mean we store scr_el3 to cm_context only once and retore it every time in el3_exit? Now I realize that scr_el3 is not the same as elr_el3 or spsr_el3, while the latter two might be changed every time CPU enters EL3; scr_el3 will not be changed in other ELs.
Is my understanding right?

Regards,
Gary

geesun, 2017-12-26:
Your understanding is correct. 

scr_el3 only access in EL3, it defines the configuration of the current Security state. It specifies:
- The Security state of EL0 and EL1, either Secure or Non-secure.
- The Execution state at lower Exception levels.
- Whether IRQ, FIQ, and External Abort interrupts are taken to EL3.

elr_el3 and spsr_el3 only access in EL3, but it may be modified when taken an exception into EL3, and this is done by hardware.

iwishguo, 2017-12-27:
Thanks. That's clear now.

Gary

closed 2017-12-27 by iwishguo