ARM-software/tf-issues #540: spin_lock may not work on some ARM64 platform when cache is disabled

iwishguo, 2017-12-28:
Hi,
I'm trying to upgrade TF to v1.4 on our platform which contains cortex A57 cores. I found the secondary cores hung at aquiring spin lock when being waken up. The issue can be fixed by enabling cache at line 178 in bl31_entrypoint.S, which is not enabled now for HW_ASSISTED_COHERENCY is 0 by default.
In ARMv8 (ARM DDI 0487B.a id033117) "B2.9.2 Exclusive access instructions and Shareable memory locations", it says global monitor may not work especially for non-cacheable memory regions. So it seems our platform does not support load/store exclusive when cache is disabled, and it is still compatible with ARMv8 specification.
If my understanding is right, then spin_lock actually depends on cacheable memory (both with MMU configured and cached enabled), and we'd better use it by cerntain condition.

Thanks,
Gary

soby-mathew, 2018-01-03:
Hi Gary,
Unless I am missing something, there is a no spin lock acquisition with caches disabled in the secondary CPU warm boot code path when HW_ASSISTED_COHERENCY is 0. 

If you referring to the lock acquisition here: https://github.com/ARM-software/arm-trusted-firmware/blob/master/lib/psci/psci_common.c#L767, this is a `bakery` [1] lock which is designed to work with varying cache states. 

The only spin_lock acquired is done with caches enabled (https://github.com/ARM-software/arm-trusted-firmware/blob/master/lib/psci/psci_on.c#L178).  The caches are enabled in `psci_do_pwrup_cache_maintenance()`.

Which one of the 2 locks mentioned above are you having trouble with?

[1] http://lamport.azurewebsites.net/pubs/bakery.pdf

iwishguo, 2018-01-16:
Thanks very much for your explanation; I found we made a mistake by hacking psci_do_pwrup_cache_maintenance so that data cache was not enabled in this function.
Cheers,
Gary

closed 2018-01-16 by iwishguo