ARM-software/tf-issues #542: Technique or Tool used in CVE-2017-5753 "code pattern" detection

vsiles, 2018-01-09:
Hi !
I'm interested in using ARM's ATF for an embedded project, and I read on
the https://developer.arm.com/support/security-update webpage that the
ATF is not affected by the Variant 1 of the Specter attack.

Did you review the code "by-hand" to convince yourself that the harmful
pattern is not present or did you use some kind of analysis tool ? I'd
like a real insurance that the code is not affected at all.

Best regards,                                                                                                                                                                                                                               
Vincent

danh-arm, 2018-01-09:
Hi

> ... I read on the https://developer.arm.com/support/security-update webpage that the
ATF is not affected by the Variant 1 of the Specter attack.

No, we never said that. To quote the [TF advisory](https://github.com/ARM-software/arm-trusted-firmware/wiki/ARM-Trusted-Firmware-Security-Advisory-TFV-6):

"At the time of writing, no vulnerable patterns have been observed in upstream TF code, therefore no workarounds have been applied or are planned."

It's possible that vulnerable patterns may be found in future.

> Did you review the code "by-hand" to convince yourself that the harmful pattern is not present or did you use some kind of analysis tool ?

Yes, we reviewed the code "by-hand". We are aware of some prototype analysis tools but these are still in development and are not yet suitable for wide distribution. Early use of these indicates many false positives, which will take significant time to process. We will continue to evaluate these tools in the coming weeks/months.

>  I'd like a real insurance that the code is not affected at all.

Sorry, we can't offer that insurance. You should perform your own review/analysis if you are concerned.

Regards

Dan.

vsiles, 2018-01-10:
Thank you for the feedback. By any chance, could you point me to some of these tools ?

closed 2018-02-22 by danh-arm