ARM-software/tf-issues #545: Question about the Workarounds for CVE-2017-5715 A9/A15

vsiles, 2018-01-30:
Hi !
In order to implement the workaround for CVE-2017-5715 in 32 bit mode (see
https://github.com/ARM-software/arm-trusted-firmware/pull/1228 and more specifically https://github.com/ARM-software/arm-trusted-firmware/pull/1228/commits/7343505d9661ab6481e99fa96d60f2a8447a4565), you installed a new MVBAR which performs some operations on `sp` instead of jumping directly to the relevant handler.

Did you choose this technique to avoid calling the workaround code from every exception handler, or is there an issue to perform a direct branch `b foobar` in the context of Spectre ?

dp-arm, 2018-01-30:
> Did you choose this technique to avoid calling the workaround code from every exception handler, or is there an issue to perform a direct branch b foobar in the context of Spectre ?

That's exactly it yes.  Doing any form of branch before applying the workaround is dangerous because of Spectre.

vsiles, 2018-01-30:
Thank you for this information, I wasn't sure "direct" branch were affected.

closed 2018-01-30 by vsiles