ARM-software/tf-issues #547: Trusted SRAM Top and Bottom

bjwrkj, 2018-02-12:
Hi,

Why we put BL1 at the top of the Trusted SRAM?
For now, BL31 at the top of the Trurted SRAM, and I tried put BL1 in top of the SRAM. 
but loading BL31 image will be hang.

log:
NOTICE:  Booting Trusted Firmware
NOTICE:  BL1: v1.4(release):v1.4-612-g37b3f21-dirty
NOTICE:  BL1: Built : 06:31:02, Feb 12 2018
NOTICE:  BL1: Booting BL2
NOTICE:  BL2: v1.4(release):v1.4-612-g37b3f21-dirty
NOTICE:  BL2: Built : 06:30:47, Feb 12 2018

Who can  tell me why ? 

Thanks a lot.
Keven

hzhuang1, 2018-02-12:
BL1 is designed for ROM. If there's no-TF ROM used in your SoC, you could enable BL2_EL3 that means running BL2 at EL3 mode. This feature is a new one.

In your case, you need to enable crash console watching what happened. I doubt that you haven't mapped the region in BL2.

bjwrkj, 2018-02-12:
Thanks for you, @hzhuang1 .

For now,  BL31 image was loaded by BL1 when used LOAD_IMAGE_V2 version. 
so mapped the region in BL2 is no effective?

Does means, in LOAD_IMAGE_V2 version image loading, can't put BL1 at the top of the Trusted SRAM?

danh-arm, 2018-02-12:
Hi

> Why we put BL1 at the top of the Trusted SRAM?

The BL1 RW data location is a platform specific choice (see BL1_RW_BASE definition). On Arm platforms, we've found putting this at the top of Trusted SRAM provides the best RAM footprint by overlaying image data that is not used concurrently. See https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/firmware-design.rst#921memory-layout-on-arm-development-platforms

> For now, BL31 at the top of the Trurted SRAM, and I tried put BL1 in top of the SRAM.
but loading BL31 image will be hang.

It sounds like you're putting BL1 RW data and BL31 at the same place, which will cause the BL1 RW data to be corrupted when you load BL31.

> For now, BL31 image was loaded by BL1 when used LOAD_IMAGE_V2 version.
so mapped the region in BL2 is no effective?

BL31 is usually loaded by BL2, not BL1. I don't really understand your question but the loading image (BL1 or BL2) must map in the target memory for all images it loads.

> Does means, in LOAD_IMAGE_V2 version image loading, can't put BL1 at the top of the Trusted SRAM?

With LOAD_IMAGE_V2, the code expects BL1 RW data to be at the top of the memory visible to BL1. This may or may not be at the top of Trusted SRAM. See https://github.com/ARM-software/arm-trusted-firmware/blob/master/bl1/bl1_main.c#L49

bjwrkj, 2018-02-23:
@danh-arm  Got it , Thanks for you.  
Happy Chinese New Year.

closed 2018-02-24 by bjwrkj