ARM-software/tf-issues #550: Need a solution to fix warnings from deprecated APIs

masahir0y, 2018-02-21:
This is related to
https://github.com/ARM-software/arm-trusted-firmware/pull/1238

@danh-arm mentioned that this should be discussed in tf-issue,
so I am opening a new issue.

> The compat layer may not be necessary for you but it is for others. It's entirely possible to remove those false positives as suggested in ARM-software/tf-issues#407.

This is possible only when no deprecated API is used on your platform.
More and more APIs are deprecated all the time, endlessly.

If ERROR_DEPRECATED is set to 1 in your build script,
it will suddenly break when the upstream code deprecates another API.

Platform maintainers must keep up with the latest API.
This is too much burden.
Some cases are trivial to migrate to, but some are not.
Setting aside Arm platforms, other vendors are not catching up.

For example, I have not been able to find time to implement MULTI_CONSOLE
because I am busy for other projects.

So, I can not set ERROR_DEPRECATED=1 even on my local code base.

The only solution is, to accept the scattered build log.
We would easily miss important warning in messy log.

> The expected usage of this is to use ERROR_DEPRECATED=0 in the default build config,
> and use ERROR_DEPRECATED=1 in a special build config. When the latter fails,
> it's an indication that the platform should do some API migration work,
> but it should not be a critical failure.

Again, ERROR_DEPRECATED=1 is possible
only when all APIs in use are up-to-date.

The platforms maintained by Arm Ltd. can make it
because most of API changes come from Arm.

OK, MULTI_CONSOLE came from Google, but
you (Dan & David) have full control of pull request.
You can prepare for it before you decide to pull it. 

Given that the difference of the number of developers, 
(see, most of contributers are from Arm)
this is not realistic for other vendors.
(it does not look so to me, at least)

>Marking deprecated APIs with a date doesn't really help platforms with migration or with API removal.
>Platforms only need to know whether they need to do migration work or not (which the existing flag enables).
>API removal will happen in 1 big break, independent to the date the API was deprecated.

I do not insist on date-driven removal.  Just for-instance solution.

Any sensible solution is OK,
but ERROR_DEPRECATE looks a violent solution to me.

As I mentioned in the commit log, the root cause of the problem is
ERROR_DEPRECATED is all-or-nothing switch.

Any old API breaks the build, so nobody (except Arm) can set
ERROR_DEPRECATED=1 for ever.

Another solution could be per-API decprecate flag.

ERROR_DEPRECATED_FOO
ERROR_DEPRECATED_BAR
...

Every time you deprecate an API, add ERROR_DEPRECATED_*

With this, no platform would break suddenly, even though
some warnings are displayed.

When platform finishes migration to the new version of feature FOO,
set ERROR_DEPRECATED_FOO to 1, to suppress annoying warnings.

Thought?

danh-arm, 2018-02-22:
>>  The compat layer may not be necessary for you but it is for others. It's entirely possible to remove those false positives as suggested in ARM-software/tf-issues#407.

> This is possible only when no deprecated API is used on your platform.
More and more APIs are deprecated all the time, endlessly.

The false positives warnings can be removed irerespective of whether you use deprecated APIs on your platform or not. But your issue seems to be with the deprecated warnings generally...

> The only solution is, to accept the scattered build log.
We would easily miss important warning in messy log.

The only warnings in the build log are these deprecated warnings. These are either due to the compatibility layers in the codebase (which I've already said we can suppress with some work) or due to the platform really using a deprecated API. All other warnings are treated as errors.

> Again, ERROR_DEPRECATED=1 is possible
only when all APIs in use are up-to-date.

Correct - that is why it should only be used in non-critical build configurations, to force errors when using deprecated APIs, which otherwise would only appear as warnings in the "scattered build log".

> The platforms maintained by Arm Ltd. can make it
because most of API changes come from Arm.

Yes, it's easier for Arm but even we can't always keep up. While its desirable for platforms to migrate, they  can do this at their own speed since we strive to maintain compatiblity wherever possible.

> OK, MULTI_CONSOLE came from Google, but
you (Dan & David) have full control of pull request.
You can prepare for it before you decide to pull it.

The Arm platforms still haven't migrated to MULTI_CONSOLE_API yet. Actually this is good example of where a new build flag really is needed, because there is no compatibility layer when this API is enabled. But note the value of this defaults to ERROR_DEPRECATED to help notify platforms that there's something to do.

> Any old API breaks the build, so nobody (except Arm) can set
ERROR_DEPRECATED=1 for ever.

Not even Arm can set this forever. That is not the intended usage of this build flag!

> Another solution could be per-API decprecate flag.
>
> ERROR_DEPRECATED_FOO
> ERROR_DEPRECATED_BAR
> ...

OK, let's explore this a bit. If ERROR_DEPRECATED_FOO is introduced to deprecate the FOO APIs, this will cause deprecated warnings in the platform until the platform is updated to use the new FOO APIs. At that point there will be no deprecated warnings, irrespective of whether the platorm sets ERROR_DEPRECATED_FOO or not. But I guess the platform could set ERROR_DEPRECATED_FOO to 1 to ensure it doesn't "go back" to the deprecated APIs (I don't know why it would). Sometime later, ERROR_DEPRECATED_BAR is introduced, and new deprecated warnings appear in the platform. At this point, the platform is in exactly the same situation as it would be if the API were deprecated using the current ERROR_DEPRECATED method. I don't see how the platform has benefited at all.

The only use case I can see for per-API deprecated flags is to be able to filter the deprecated warnings (or build errors if the flag is on) to a smaller, presumably more manageable, subset of the total deprecated warnings/errors. Is this the real reason for your request? Even then, there would be no real need to set these flags in the platform make file, and there would still be a need for an overall ERROR_DEPRECATED flag that could set all these per-API flags to 1.

masahir0y, 2018-02-27:
> OK, let's explore this a bit. If ERROR_DEPRECATED_FOO is introduced to deprecate the FOO APIs, this will cause deprecated warnings in the platform until the platform is updated to use the new FOO [...]

Let me clarify my point.  When I compile my platform with PLAT=uniphier, I see several warnings.

    plat/socionext/uniphier/uniphier_bl31_setup.c:86:2: warning: ‘console_uninit’ is deprecated [-Wdeprecated-declarations]

I think this is related to MULTI_CONSOLE.  I am totally fine with this warning.  I have not finished MULTI_CONSOLE migration.  It is good to see it as a reminder of my TODO item.

However, I also see these.

    bl31/bl31_context_mgmt.c: In function ‘cm_get_context_by_mpidr’:
    bl31/bl31_context_mgmt.c:82:2: warning: ‘platform_get_core_pos’ is deprecated [-Wdeprecated-declarations]

and

    plat/common/aarch64/plat_common.c: In function ‘plat_get_syscnt_freq2’:
    plat/common/aarch64/plat_common.c:67:2: warning: ‘plat_get_syscnt_freq’ is deprecated [-Wdeprecated-declarations]

This is weird because I finished the migration.  Why should I be bothered about what I did correctly?

danh-arm, 2018-02-27:
> This is weird because I finished the migration. Why should I be bothered about what I did correctly?

Yes, I agree! We can get rid of these (as previously stated). It's just that no-ones bothered to do this. I'm going to submit patch to make these go away. Then hopefully you won't need these alternative flags.

closed 2018-03-04 by davidcunado-arm