ARM-software/tf-issues #552: Assert when trying to run ATF with FVP foundation model

ruchi393, 2018-02-22:
Hi,

I am following instructions in the ATF user guide for running ATF via FVP foundation model. 

I get the following assert while doing so :
**ASSERT: lib/xlat_tables_v2/aarch64/xlat_tables_arch.c:64**

Can someone help on what could be going wrong over here ?

FVP model being used -
Foundation_Platform  --version
ARM V8 Foundation Platformr0p0 (platform build 11.2.33)
Copyright 2017 ARM Limited.
All Rights Reserved.

ATF images being built from following commit ID

commit 15e5958560e9d31e7357e3a0ada2289e78758839
Merge: e47a832 fb750ee
Author: davidcunado-arm <david.cunado@arm.com>
Date:   Mon Feb 12 17:52:08 2018 +0800

    Merge pull request #1256 from jeenu-arm/tsp-ehf
    
    TSP changes for EHF

Thanks,
Ruchika

davidcunado-arm, 2018-02-22:
@ruchi393 
What build command / options are you using?

antonio-nino-diaz-arm, 2018-02-22:
If you aren't specifying the version of the architecture that the model is simulating, it defaults to the highest one it can emulate. The field `PARANGE` of register `ID_AA64MMFR0_EL1` gained a new value in 8.2. The model uses the max value that `PARANGE` supports on the architecture. If the model is running at version 8.2 or higher, it will use the new value introduced in 8.2. If you don't compile the TF for version 8.2 or higher, it won't know about this extra value and you will hit this assert.

Your options are:
- Run the foundation model with `--arm-v8.0` or `--arm-v8.1`.
- Compile the firmware with `ARM_ARCH_MAJOR=8` and `ARM_ARCH_MINOR=2` or higher.

danh-arm, 2018-02-22:
Hi @antonio-nino-diaz-arm 

If what you say is correct, then the model parameters in the user guide are wrong. This should be fixed!

ruchi393, 2018-02-23:
Hi,

Thanks for the help. The assert goes away if --arm-v8.0 is added when launching FVP.  

As per the user guide, I have picked up imaged from Linaro release. I used workspace_1801.py  from the link  https://community.arm.com/dev-platforms/w/docs/226/old-linaro-release-note. I have tried both pre-built as well as  build options. However, I don't see a uboot or linux prompt coming up. (This pre built release also gives the same assert error)

Can someone please help point to the images which I should pick up from Linaro which would work with this FVP model.

Regards,
Ruchika

antonio-nino-diaz-arm, 2018-02-23:
I'll update the instructions of the User Guide, @danh-arm.

We are using Linaro Release 17.10, @ruchi393, could you see if that one works for you, at least?

closed 2018-09-28 by antonio-nino-diaz-arm