ARM-software/tf-issues #555: Query on Cryptographic Module - Authentication framework

ruchi393, 2018-02-26:
Hi,

I am exploring on how to use the interfaces defined by authentication and cryptographic module in ATF for my platform defines images (I will be using IMG_PLAT).

For signatures, the platform I am working on uses RSA signature with SHA-256. If I look at the interface definition for signature in the CM module given below, only a single data pointer is provided  in the interface. Here in the function below , the assumption is that all the data on which hash needs to be calculated for verifying with signature is present in contiguous memory. However there may be scenarios where data is present in dis-contiguous memory.

int (*verify_signature)(void *data_ptr, unsigned int data_len,
                        void *sig_ptr, unsigned int sig_len,
                        void *sig_alg, unsigned int sig_alg_len,
                        void *pk_ptr, unsigned int pk_len);

I would like to extend the function  to take multiple data segments. The crypto library can then invoke the hash function accordingly.

Does anyone see any issues with above approach ?

Regards,
Ruchika

danh-arm, 2018-02-26:
> Does anyone see any issues with above approach ?

Your requirement sounds reasonable. I suggest when you have a proposed new API (or high level design), you post it here rather than raising a complete Pull Request in the first instance. That might save time in review.

I imagine it might be useful to define some kind of generic scatter/gather API, which is then used in these crypto module APIs.

dp-arm, 2018-02-27:
Just a thought, but maybe it is possible to break this into multiple functions and have an init function, an update function and a finalize function.  So the caller will typically call init(), update() (potentially multiple times) and then finalize().  This means that if you don't have all the data upfront, you can just call update() on each buffer as it becomes available.

I would look at what APIs existing crypto libraries have for this kind of stuff.

ruchi393, 2018-02-27:
Usually the cryptographic libraries maintain a context which is library specific with init, update and finalize function. This would vary with the crypto module you integrate. What I understand is that here the intention is to keep the authentication module independent of the crypto library. So the idea would be to pass the multiple data entries to the crypto library and let it decide on how to handle it.

Given below a very high level proposal below. I am new to the authentication framework in ATF , so I might have missed out on few things. Kindly review and let me know if I am going in the right direction

An auth_param_type can be added which points to AUTH_PARAM_SG_DATA which is used in both hash and verify function instead of RAW_DATA. 

When authentication framework returns AUTH_PARAM_SG_DATA data , it can fill the discontiguous data entries (sg entries) in tabular format and return the table pointer in *ptr field of auth_param_data_desc_s. The len field would indicate the number of entries.

int verify_signature(void *sg_tbl, unsigned int num_entries,
                     void *sig_ptr, unsigned int sig_len,
                     void *sig_alg, unsigned int sig_alg_len,
                     void *pk_ptr, unsigned int pk_len);

struct sg_entry {
        void *data;
       unsigned int len;
};

SG table is an array of these entries.

The cryptographic module can read the number of entries to figure out whether there is single data field or multiple and accordingly  call single hash or hash init/update and finalize function.

typedef struct auth_method_param_sig_s {
        auth_param_type_desc_t *pk;     /* Public key */
        auth_param_type_desc_t *sig;    /* Signature to check */
        auth_param_type_desc_t *alg;    /* Signature algorithm */
        auth_param_type_desc_t *data;   /* Data signed */
} auth_method_param_sig_t;

In signature authentication method. the data descriptor would now be of type AUTH_PARAM_SG_DATA .

Regards,
Ruchika

ruchi393, 2018-02-27:
2nd proposal :
One more method can be that hash is calculated by the image parser library in case of dis-contiguous data and verify function is changed to include a type. This type field can indicate what type of data is passed to verify_Signature function if it is hash or raw data. This how openssl does it.

int verify_signature(int type , void *data, unsigned int len,
void *sig_ptr, unsigned int sig_len,
void *sig_alg, unsigned int sig_alg_len,
void *pk_ptr, unsigned int pk_len);

Please note that in this approach also, the parser library needs to calculate hash over discontiguous data, so some function would need to be exposed from cryptographic module which takes in these entries, calculates hash and returns back to the parser library.

ruchi393, 2018-03-09:
Hi,

Please share any feedback on the above suggested approaches.

Regards,
Ruchika

dp-arm, 2018-03-09:
@ruchi393 Can I ask how you bring the image into memory in the first place?  Do you have a custom I/O layer implementation?  The current code in TF loads images in one contiguous chunk.

ruchi393, 2018-03-13:
@dp-arm , No changes are planned in io later. Instead of certificate or RAW image, on my platform I would have a IMAGE_PLAT (This contains a header prepended with a BL31/other images). During the authentication check on this platform specific image, hash needs to be calculated from certain portions in the header and the appended image. 

One option can be that I fill a SG table with these chunks of memory and pass that in verify signature.
or
I can calculate the hash of the data entries in the check_integrity function. In verify_signature, I can use hash instead of data entries. This can be specified via proposed type in verify_signature function.

dp-arm, 2018-03-26:
@ruchi393 The second approach sounds good to me.

ruchi393, 2018-03-27:
@danh-arm. I can have my crypto library expose this function. Since these are generic cryptographic  functions and we already have a crypto module in ATF, my opinion was that it would be better to expose these out of these crypto module and let the image parser library use it.

danh-arm, 2018-03-28:
Hi @ruchi393. There is only value in abstracting these new functions in the generic crypto module if the functions actually get used by some other part of the generic authentication framework (e.g. the authentication or image parser modules). Could some of the functionality in your image parser library be brought up to the generic level? Maybe you could provide a standard helper image parser library that wraps other image parser libraries?

dp-arm, 2018-04-16:
@ruchi393 ping

ruchi393, 2018-04-17:
Currently I don't see any use of these crypto functions in the generic authentication framework in the approach where image parser takes care of this and returns hash. For the helper library part of image parser library, I haven't encountered any commonalities as of now. If I do, I will submit a patch for the same,

closed 2018-04-17 by ruchi393