ARM-software/tf-issues #563: Update doc for ARM TF on ARMv7 QEMU

soby-mathew, 2018-03-08:
The current [QEMU documentation](https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/plat/qemu.rst) only refers to build and run instructions for ARM64 QEMU. It will be very useful to have this documentation updated for ARMv7 QEMU as well.

soby-mathew, 2018-03-08:
@etienne-lms , @jenswi-linaro ,  We are also interested in running the Trusted Board Boot with Optee as BL32 on ARMv7 QEMU. If you could provide instructions for the same, it will be very helpful.

etienne-lms, 2018-03-12:
Hi @soby-mathew, the optee part (support for being booted from arm-tf) is still under review, hence not yet in the official op-tee repo. P-R https://github.com/OP-TEE/optee_os/pull/2138 is a proposal but needs to be reworked. I'll let you know when we have something merged or about to be.

etienne-lms, 2018-10-05:
Hi, 
Actually, the OP-TEE delivery now integrates a Qemu setup that boots OP-TEE/Linux on a quad cortex-a15 using TrustedFirmware as Armv7 bootloader. The easiest way is to retrieve the OP-TEE default delivery, see [get-and-build-the-(optee)-solution](https://github.com/OP-TEE/build#7-get-and-build-the-solution), in few commands:
```
$ mkdir -p $HOME/optee-test && cd $HOME/optee-test && \
  repo init -u https://github.com/OP-TEE/manifest.git -m default_stable.xml && \
  repo sync
```
to get the source, and following to build (and run Qemu):
```
$ cd build && make toolchains && make all
$ make run-only
```

etienne-lms, 2018-10-05:
ok, i will propose something...