ARM-software/tf-issues #564: Userdata not found during flashing ARM-TF + Android for Hikey960

dhimanchakraborty, 2018-03-08:
While flashing the hikey960 board to ARM-TF+Android, I am getting “Couldn’t erase image: Not Found” error.

```
FastbootTransportUsbRequestReceive, 255, BufferSize:24576
   24576 /    24576 bytes downloaded (100%)
Flashing partition ptable
Done.
Fastboot platform: check for partition-type:xloader
Downloading 154752 bytes
#FastbootTransportUsbRequestReceive, 255, BufferSize:154752
  154752 /   154752 bytes downloaded (100%)
Flashing partition xloader
Done.
Fastboot platform: check for partition-type:fastboot
Downloading 1179648 bytes
#FastbootTransportUsbRequestReceive, 255, BufferSize:1179648
 1179648 /  1179648 bytes downloaded (100%)
Flashing partition fastboot
Done.
Fastboot platform: check for partition-type:fip
Downloading 1531904 bytes
#FastbootTransportUsbRequestReceive, 255, BufferSize:1531904
 1531904 /  1531904 bytes downloaded (100%)
Flashing partition fip
Done.
Fastboot platform: check for partition-type:boot
Fastboot platform: check for partition-type:boot
Downloading 10070016 bytes
#FastbootTransportUsbRequestReceive, 255, BufferSize:10070016
10070016 / 10070016 bytes downloaded (100%)
Flashing partition boot
Done.
Fastboot platform: check for partition-type:cache
Erasing partition cache
Downloading 57492 bytes
#FastbootTransportUsbRequestReceive, 255, BufferSize:57492
   57492 /    57492 bytes downloaded (100%)
Flashing partition cache
Done.
Fastboot platform: check for partition-type:system
Erasing partition system
Downloading 134213804 bytes
#FastbootTransportUsbRequestReceive, 255, BufferSize:134213804
134213804 / 134213804 bytes downloaded (100%)
Flashing partition system
Done.
Downloading 134213696 bytes
#FastbootTransportUsbRequestReceive, 255, BufferSize:134213696
134213696 / 134213696 bytes downloaded (100%)
Flashing partition system
Done.
Downloading 134213720 bytes
#FastbootTransportUsbRequestReceive, 255, BufferSize:134213720
134213720 / 134213720 bytes downloaded (100%)
Flashing partition system
Done.
Downloading 134213696 bytes
#FastbootTransportUsbRequestReceive, 255, BufferSize:134213696
134213696 / 134213696 bytes downloaded (100%)
Flashing partition system
Done.
Downloading 134213720 bytes
#FastbootTransportUsbRequestReceive, 255, BufferSize:134213720
134213720 / 134213720 bytes downloaded (100%)
Flashing partition system
Done.
Downloading 134213696 bytes
#FastbootTransportUsbRequestReceive, 255, BufferSize:134213696
134213696 / 134213696 bytes downloaded (100%)
Flashing partition system
Done.
Downloading 134213720 bytes
#FastbootTransportUsbRequestReceive, 255, BufferSize:134213720
134213720 / 134213720 bytes downloaded (100%)
Flashing partition system
Done.
Downloading 35553440 bytes
#FastbootTransportUsbRequestReceive, 255, BufferSize:35553440
35553440 / 35553440 bytes downloaded (100%)
Flashing partition system
Done.
Fastboot platform: check for partition-type:userdata
Erasing partition userdata
Couldn't erase image:  Not Found
```
I am using prm_ptable.img, sec-xloader.img, l-loader.img, fip.img, boot.img, cache.img etc as told in the github page for flashing hikey960 for ARM-TF.

And If I ignore this error and try to boot the Hikey960 I am getting the below error,

```
[Bds]Booting Boot from SD
[Bds]Booting Grub
[Bds]Booting Android Boot
add-symbol-file /home/dhiman/colossus5/simAsTpm/edk2/Build/HiKey960/DEBUG_GCC5/AARCH64/EmbeddedPkg/Application/AndroidBoot/AndroidBootApp/DEBUG/AndroidBootApp.dll 0xB9B07000
Loading driver at 0x000B9B06000 EntryPoint=0x000B9B07000 AndroidBootApp.efi
The kernel image is not raw format: Invalid Parameter
ERROR: Device Tree header not valid (err:-9)
Failed to get attached FDT from the end of raw kernel: Invalid Parameter
Failed to boot from partition: Invalid Parameter
Error: Image at 000B9B06000 start failed: Invalid Parameter
```
I guess the problem is with the ptable I am using. I tried with hisi-ptable.img (generated from android hikey960-userdebug build) which yields no better result. What could be the problem?

hzhuang1, 2018-03-09:
@dhimanchakraborty Please help to create bug in http://bugs.96boards.org. Then it could be easy to track. By the way, I need to know more in details. Could you help to list the uefi build number if you're using prebuilt uefi? If you're building the firmware by yourself, could you help to list all commits that you're using now?

hzhuang1, 2018-03-09:
> The kernel image is not raw format: Invalid Parameter

It's the compressed kernel image.

> ERROR: Device Tree header not valid (err:-9)

It can't find DTB image. You're using original boot image. But it's a bit different when we use UEFI. You need to bind kernel and DTB together.

 Yea, we have a few pending patches for the UEFI transition:
 https://android-review.googlesource.com/q/topic:%22hikey960-uefi-transition%22+(status:open%20OR%20status:merged)
 
 You need config changes to build the kernel as a Image.gz-dtb
 https://android-review.googlesource.com/c/kernel/hikey-linaro/+/501533
 
 And some BoardConfig logic to pick up the new kernel name:
 https://android-review.googlesource.com/c/device/linaro/hikey/+/458741

dhimanchakraborty, 2018-03-09:
@hzhuang1 I am building my own. Commit details are below,

l-loader : testing/hikey960_v1.2
arm-trusted-firmware : integration
OpenPlatfomrPkg : testing/hikey960_v1.3.4
uefi-tools : master (pulled on August 31 2017)
edk2 : testing/hikey960_v2.5

prm_ptable.img generated from l-loader having no userdata

Number  Start (sector)    End (sector)  Size       Code  Name                                                                                                                                                                                                                                                           
   1             256             511   1024.0 KiB  0700  vrl                                                                                                                                                                                                                                                               
   2             512            3583   12.0 MiB    0700  fastboot                                                                                                                                                                                                                                                          
   3            3584            5119   6.0 MiB     0700  nvme                                                                                                                                                                                                                                                              
   4            5120            8191   12.0 MiB    8300  fip                                                                                                                                                                                                                                                               
   5            8192           73727   256.0 MiB   0700  cache                                                                                                                                                                                                                                                             
   6           73728           73983   1024.0 KiB  0700  fw_lpm3                                                                                                                                                                                                                                                           
   7           73984           90367   64.0 MiB    EF00  boot                                                                                                                                                                                                                                                              
   8           90368           94463   16.0 MiB    0700  dts                                                                                                                                                                                                                                                               
   9           94464           94975   2.0 MiB     0700  trustfirmware                                                                                                                                                                                                                                                     
  10           94976         1295103   4.6 GiB     8300  system                                                                                                                                                                                                                                                            
  11         1295104         1495807   784.0 MiB   0700  vendor                                                                                                                                                                                                                                                            
  12         1495808         1496063   1024.0 KiB  0700  reserved

vchong, 2018-03-09:
What command did you use to build the ptable? Do you see `ptable-aosp32.img` in your `l-loader` dir? If yes then try using that.

dhimanchakraborty, 2018-03-12:
Hello Victor (@vchong)
Hello,

I have used the commands from the github page          https://github.com/ARM-software/arm-trustedfirmware/blob/master/docs/plat/hikey960.rst

I do not see ptable-aosp32.img in l-loader folder. Below is the list of ptable I have in l-loader folder.

-rw-r--r-- 1 chakraborty chakraborty 24576 Mär  9 15:24 ptable-aosp-32g.img
-rw-r--r-- 1 chakraborty chakraborty 24576 Mär  9 15:25 ptable-aosp-32g-spare.img
-rw-r--r-- 1 chakraborty chakraborty 24576 Mär  9 15:25 prm_ptable.img
-rw-r--r-- 1 chakraborty chakraborty 20480 Mär  9 15:25 sec_ptable.img
-rw-r--r-- 1 chakraborty chakraborty 24576 Mär  9 15:25 ptable-linux-32g.img

I can try with ptable-aosp-32g.img instead.

dhimanchakraborty, 2018-03-14:
ptable-aosp-32g.img solves the problem with userdata partition.

closed 2018-03-15 by hzhuang1