ARM-software/tf-issues #565: "The kernel image is not raw format: Invalid Parameter" flashing ARM-TF+Android for Hikey960

dhimanchakraborty, 2018-03-14:
I have the master branch of android **https://android.googlesource.com/platform/manifest -b master**
I have the hikey kernel from google **https://android.googlesource.com/kernel/hikey-linaro**
I am building my own ARM-TF. Commit details are below,
```
l-loader : testing/hikey960_v1.2
arm-trusted-firmware : integration
OpenPlatfomrPkg : testing/hikey960_v1.3.4
uefi-tools : master (pulled on August 31 2017)
edk2 : testing/hikey960_v2.5
```
Added dtb image creation entry in hikey_defconfig following **https://android-review.googlesource.com/#/c/kernel/hikey-linaro/+/501533/**
After compiling the kernel copied  **/hikeyKernel/hikey-linaro/arch/arm64/boot/dts/hisilicon/hi3660-hikey960.dtb** to android as **hi3660-hikey960.dtb-4.9** 
and **/android/hikeyKernel/hikey-linaro/arch/arm64/boot/Image.gz-dtb** as **Image.gz-hikey960-4.9** to android. Then compiled android. Then after flushing hikey960 with **ptable-aosp-32g.img, sec_xloader.img, l-loader.bin, fip.bin, boot.img, dt.img, system.img, cache.img, userdata.img** getting **kernel image is not in raw format**.

```
.[Bds]Booting Boot from SD
[Bds]Booting Grub
[Bds]Booting Android Boot
add-symbol-file /home/dhiman/colossus5/simAsTpm/edk2/Build/HiKey960/DEBUG_GCC5/AARCH64/EmbeddedPkg/Application/AndroidBoot/AndroidBootApp/DEBUG/AndroidBootApp.dll 0xB9B07000
Loading driver at 0x000B9B06000 EntryPoint=0x000B9B07000 AndroidBootApp.efi
The kernel image is not raw format: Invalid Parameter
ERROR: Device Tree header not valid (err:-9)
Failed to get attached FDT from the end of raw kernel: Invalid Parameter
Failed to boot from partition: Invalid Parameter
Error: Image at 000B9B06000 start failed: Invalid Parameter
```
also getting the below error which I am not understanding
```
Ftw: Workspace or Spare block does not exist!
Error: Image at 000BF034000 start failed: Invalid Parameter
```
Q1. What could be the problem?
Q2. Is there anyway to make the console logging more verbose in Hikey960?

hzhuang1, 2018-03-15:
How do you generate boot.img? It seems there's something wrong in the boot.img.

Before flushing boot image, you could try to use "fastboot boot boot.img" to verify whether the boot image is well.

You can use "abootimg" tool to unpack everything from the boot image, and replace kernel/zImage by your kernel image with DTB.  Then pack it again to get a new boot.img. Or you can use "bootimg-960.cfg" as a reference in l-loader to generate a new boot image.

For the Ftw error message, just ignore it. We didn't use it at all.

dhimanchakraborty, 2018-03-15:
I am using the below process to get the dtb and Image.gz from hikey kernel to android and building the bootimage

```
Building the kernel
    Run the following commands:
    git clone https://android.googlesource.com/kernel/hikey-linaro
    cd hikey-linaro
    git checkout -b android-hikey-linaro-4.9 origin/android-hikey-linaro-4.9
    make ARCH=arm64 hikey960_defconfig
    make ARCH=arm64 CROSS_COMPILE=aarch64-linux-android- -j24

    Update the kernel in the boot image.
        Copy hi3660-hikey960.dtb (arch/arm64/boot/dts/hisilicon/hi3660-hikey960.dtb) to the hikey-kernel directory as file: hi3660-hikey960.dtb-4.9
        Copy the Image file (arch/arm64/boot/Image.gz) to the hikey-kernel directory as file: Image.gz-hikey960-4.9

    Make the boot image:
    make bootimage -j24
```
https://source.android.com/setup/devices - Link I am using

dhimanchakraborty, 2018-03-15:
**Before aboot**
Below is the details of the boot.img
```
dhiman@android-class-3-> ./abootimg -i ../fastboot/boot.img                                                                                                                                                                                                                                                              
Android Boot Image Info:
 
* file name = ../fastboot/boot.img
 
* image size = 10096640 bytes (9.63 MB)
  page size  = 2048 bytes
 
* Boot Name = ""
 
* kernel size       = 8689979 bytes (8.29 MB)
  ramdisk size      = 1402662 bytes (1.34 MB)
 
* load addresses:
  kernel:       0x00080000
  ramdisk:      0x07c00000
  tags:         0x07a00000
 
* cmdline = androidboot.hardware=hikey960 console=ttyFIQ0 androidboot.console=ttyFIQ0 firmware_class.path=/system/etc/firmware loglevel=15 buildvariant=userdebug
  
* id = 0x1febc331 0x13f208e6 0x75f7b908 0x7f439f18 0xded40668 0x00000000 0x00000000 0x00000000
```
Then I tried to update the boot.img file with the kernel dtb file

`./abootimg -u ../fastboot/boot.img -k hi3660-hikey960.dtb`

After aboot

```
dhiman@android-class-3-> ./abootimg -i ../fastboot/boot.img

Android Boot Image Info:

* file name = ../fastboot/boot.img

* image size = 10096640 bytes (9.63 MB)
  page size  = 2048 bytes

* Boot Name = ""

* kernel size       = 56961 bytes (0.05 MB)
  ramdisk size      = 1402662 bytes (1.34 MB)

* load addresses:
  kernel:       0x00080000
  ramdisk:      0x07c00000
  tags:         0x07a00000

* cmdline = androidboot.hardware=hikey960 console=ttyFIQ0 androidboot.console=ttyFIQ0 firmware_class.path=/system/etc/firmware loglevel=15 buildvariant=userdebug

* id = 0x1febc331 0x13f208e6 0x75f7b908 0x7f439f18 0xded40668 0x00000000 0x00000000 0x00000000

```
Still getting the same error.

```
Loading driver at 0x000B9B06000 EntryPoint=0x000B9B07000 AndroidBootApp.efi                                                        
The kernel image is not raw format: Invalid Parameter                                                                              
Failed to uncompress kernel with gzip format: Invalid Parameter                                                                    
Failed to boot from partition: Invalid Parameter
Error: Image at 000B9B06000 start failed: Invalid Parameter
```
Ran the command : sudo fastboot boot boot.img
got below error

```
Downloading 10096640 bytes
#FastbootTransportUsbRequestReceive, 255, BufferSize:10096640
10096640 / 10096640 bytes downloaded (100%)
Booting downloaded image
The kernel is not raw format.
Failed to uncompress kernel with gzip format: Invalid Parameter
Failed to boot from RAM: Invalid Parameter
```
Do you have any other idea on generating boot image for android other than specified here https://source.android.com/setup/devices

vchong, 2018-03-15:
Without looking into this in detail, you can try referring to https://github.com/96boards-hikey/l-loader/blob/testing/hikey960_v1.2/cc to see how the kernel and boot image are built.

danh-arm, 2018-07-26:
Closing, as this thread seems to have ended

closed 2018-07-26 by danh-arm