ARM-software/tf-issues #566: Do ARMv8.2 CPUs needs to do stack maintenance in power down sequence?

leon-chen-mtk, 2018-03-16:
Hi,
We encounter a problem with ARMv8.2 CPU power down sequence with HW_ASSISTED_COHERENCY option enabled.

[CPU]
Cortex A55 and Cortex A75

[ARM-Trusted-firmware version]
ARM-Trusted-firmware master branch tag v1.4, with HW_ASSISTED_COHERENCY enabled.

In CPU idle power on flow, bl31_warm_entrypoint->bl31_plat_enable_mmu->enable_mmu_el3->enable_mmu_arch->enable_mmu_internal_el3.
There is a chance that CPU use stale cache content in L3: CPU power on and push stack to external memory entering enable_mmu_internal_el3() with MMU disabled.
After MMU is enabled(with C bit disabled) in enable_mmu_internal_el3(), CPU pop stack and read stale stack content from L3 cache if cache hit, then pop wrong return address from the stale stack.

From A55 and A75 TRM Power down sequence, CPU will disable and flush L1, L2 to L3 without SW instructions.

So if L3 still alive during a CPU power off and then power on, CPU will use stale stack from L3 after MMU is enabled.

Should ARM-Trusted-firmware need to do stack maintenance before CPU power down in psci_do_pwrdown_sequence() in HW_ASSISTED_COHERENCY enabled case?

Best Regards,
Leon

jeenu-arm, 2018-03-16:
So, I take the gist of the problem is that, during the power up path, the CPU it starts reading stale data after the MMU is turned on.

Considering the fact that `SCTLR.C` remains disabled both before and after turning on MMU, it's a bit surprising that just enabling the MMU alone changes the cache visibility. During the power up path, given that the CPU only reads what it had written back (in this case, it pops back only those values that had been pushed after the CPU had been powered up), it's unlikely that this is related to anything that was done during power down path altogether.

To narrow it further, it'd be worth testing by deferring turning MMU on until the stack is complete unwound, like where the `SCTLR.C` is enabled right now.

Could you re-test by applying the patch below, and tells us whether there's any change in behaviour?

```diff
diff --git a/bl31/aarch64/bl31_entrypoint.S b/bl31/aarch64/bl31_entrypoint.S
index 0d1077cb..e4055aed 100644
--- a/bl31/aarch64/bl31_entrypoint.S
+++ b/bl31/aarch64/bl31_entrypoint.S
@@ -170,12 +170,13 @@ func bl31_warm_entrypoint
 	 * enter coherency (as CPUs already are); and there's no reason to have
 	 * caches disabled either.
 	 */
-	mov	x0, #DISABLE_DCACHE
+	mov	x0, #(DISABLE_DCACHE | DISABLE_MMU)
 	bl	bl31_plat_enable_mmu
 
 #if HW_ASSISTED_COHERENCY || WARMBOOT_ENABLE_DCACHE_EARLY
 	mrs	x0, sctlr_el3
 	orr	x0, x0, #SCTLR_C_BIT
+	orr	x0, x0, #SCTLR_M_BIT
 	msr	sctlr_el3, x0
 	isb
 #endif
diff --git a/include/lib/xlat_tables/xlat_mmu_helpers.h b/include/lib/xlat_tables/xlat_mmu_helpers.h
index d83d7640..a70dddf1 100644
--- a/include/lib/xlat_tables/xlat_mmu_helpers.h
+++ b/include/lib/xlat_tables/xlat_mmu_helpers.h
@@ -19,6 +19,7 @@
  * setting the bit SCTLR_ELx.C.
  */
 #define DISABLE_DCACHE			(U(1) << 0)
+#define DISABLE_MMU			(U(1) << 1)
 
 /*
  * Mark the translation tables as non-cacheable for the MMU table walker, which
diff --git a/lib/xlat_tables_v2/aarch64/xlat_tables_arch.c b/lib/xlat_tables_v2/aarch64/xlat_tables_arch.c
index ffbc665f..1647c397 100644
--- a/lib/xlat_tables_v2/aarch64/xlat_tables_arch.c
+++ b/lib/xlat_tables_v2/aarch64/xlat_tables_arch.c
@@ -186,7 +186,11 @@ int xlat_arch_current_el(void)
 		dsbish();						\
 		isb();							\
 									\
-		sctlr |= SCTLR_WXN_BIT | SCTLR_M_BIT;			\
+		sctlr |= SCTLR_WXN_BIT;					\
+		if (flags & DISABLE_MMU)				\
+			sctlr &= ~SCTLR_M_BIT;				\
+		else							\
+			sctlr |= SCTLR_M_BIT;				\
 		if (flags & DISABLE_DCACHE)				\
 			sctlr &= ~SCTLR_C_BIT;				\
 		else							\
```

leon-chen-mtk, 2018-03-20:
Hi Jeenu,
The given patch didn't work.
Even M bit is enabled after CPU return from C function, CPU still get stale stack value while the C function stack pop.

I have done some experiments in order to know cache visibility, it seems that cache visibiliy is nothing to do with MMU is enabled or not.  Data cache can still be hit if C/M bit is off.

[Experiment]
1. Before a CPU power off, fill this cpu's stack with fixed pattern(0x5a5a5a505a5a5a50)
2. In the every beginning of CPU power on(before enter bl31_plat_enable_mmu function), write two 8 byte with fixed pattern(0x0000000000001234, 0x0000000000005678) in each step, then read the
modified address immediately to see the value is expected or not(should be 0x0000000000001234, 0x0000000000005678)
3. If not, then do assertion.

The assertion is always triggered.
It seems that even data read cache hit, date write to the same address will go to external memory.

The read/write not symmetric cause this stack stale problem(even M bit is off).

There are two questions need to be answered:
1. From TRM we know C bit controls cacheability, but it seems not?
2. It seems that read/write is not symmetric if the access addresses is cache hit if C/M bit is off?

Not sure my assumptions are correct, please help to clarify them, thank you.

My test code as shown below:
diff --git a/v1.4/bl31/aarch64/bl31_entrypoint.S b/v1.4/bl31/aarch64/bl31_entrypoint.S
index 419927d8e..cba51e929 100644
--- a/v1.4/bl31/aarch64/bl31_entrypoint.S
+++ b/v1.4/bl31/aarch64/bl31_entrypoint.S
@@ -10,7 +10,7 @@
 #include <pmf_asm_macros.S>
 #include <runtime_instr.h>
 #include <xlat_tables_defs.h>
-
+#include <platform_def.h>
 	.globl	bl31_entrypoint
 	.globl	bl31_warm_entrypoint
 
@@ -172,6 +172,27 @@ func bl31_warm_entrypoint
 	 * enter coherency (as CPUs already are); and there's no reason to have
 	 * caches disabled either.
 	 */
+	sub	x4, sp, #PLATFORM_STACK_SIZE
+	mov x5, sp
+2:
+#if 1
+	 mov	x0, #0x1234
+	 mov	x1, #0x5678
+	 stp x0, x1, [sp, #-0x10]!
+	 ldp x2, x3, [sp]
+	 cmp x0, x2
+	 adr	x30, foo
+	 bne do_panic
+foo:
+	cmp x1, x3
+	adr	x30, boo
+	 bne do_panic
+boo:
+	mov x6, sp
+	cmp	x4, x6
+	bne	2b
+#endif
+	mov sp, x5
 	mov	x0, #DISABLE_DCACHE
 	bl	bl31_plat_enable_mmu
 
diff --git a/v1.4/lib/psci/aarch64/psci_helpers.S b/v1.4/lib/psci/aarch64/psci_helpers.S
index afe21ebe4..4006c6f5d 100644
--- a/v1.4/lib/psci/aarch64/psci_helpers.S
+++ b/v1.4/lib/psci/aarch64/psci_helpers.S
@@ -127,6 +127,14 @@ endfunc psci_do_pwrup_cache_maintenance
  * -----------------------------------------------------------------------
  */
 func psci_power_down_wfi
+	bl	plat_get_my_stack
+	ldr	x1, =0x5a5a5a505a5a5a50
+	mov x3, #0x0
+1:
+	str	x1, [x0, #-0x8]!
+	add	x3, x3, #8
+	cmp	x3, #PLATFORM_STACK_SIZE
+	bne	1b
 	dsb	sy		// ensure write buffer empty
 	wfi
 	no_ret	plat_panic_handler

soby-mathew, 2018-03-20:
@leon-chen-mtk ,

> The assertion is always triggered.
It seems that even data read cache hit, date write to the same address will go to external memory.
The read/write not symmetric cause this stack stale problem(even M bit is off).

I am wondering why this assymetry doesn't happen during cold boot ? This is problematic if true. This would mean all memory access (including stack) are non-coherent between the read and write till caches are enabled on such platforms. We would need to check this further.

Meanwhile, could you try enabling the `C` bit earlier and see the behaviour ? 
```diff
diff --git a/bl31/aarch64/bl31_entrypoint.S b/bl31/aarch64/bl31_entrypoint.S
index 0d1077c..f4dd28d 100644
--- a/bl31/aarch64/bl31_entrypoint.S
+++ b/bl31/aarch64/bl31_entrypoint.S
@@ -154,6 +154,13 @@ func bl31_warm_entrypoint
                _init_c_runtime=0                               \
                _exception_vectors=runtime_exceptions
 
+#if HW_ASSISTED_COHERENCY || WARMBOOT_ENABLE_DCACHE_EARLY
+       mrs     x0, sctlr_el3
+       orr     x0, x0, #SCTLR_C_BIT
+       msr     sctlr_el3, x0
+       isb
+#endif
+
        /*
         * We're about to enable MMU and participate in PSCI state coordination.
         *
@@ -170,16 +177,9 @@ func bl31_warm_entrypoint
         * enter coherency (as CPUs already are); and there's no reason to have
         * caches disabled either.
         */
-       mov     x0, #DISABLE_DCACHE
+       mov     x0, #0
        bl      bl31_plat_enable_mmu
 
-#if HW_ASSISTED_COHERENCY || WARMBOOT_ENABLE_DCACHE_EARLY
-       mrs     x0, sctlr_el3
-       orr     x0, x0, #SCTLR_C_BIT
-       msr     sctlr_el3, x0
-       isb
-#endif
-
```

leon-chen-mtk, 2018-03-21:
Hi Soby,
Sorry that I made the wrong description.
It is not "always assert" in each time CPU warm reset(power on), but it is very easy to reproduce(because the frequency of kernel thread idle and wake up is high)

After discussing with our digital designer, the C bit controls cacheability, no doubt.
I have made the wrong assumption.

However, we think there is one case may lead to the weird issue we encountered:
1. One CPU power off, HW do the stack cache maintenance and the latest data keep in L3. Other CPUs may live or power off, it depends on kernel's loading
2. While this CPU power on, during this CPU push stack to external memory and before this CPU pop stack(in enable_mmu_internal_el3() function), other CPUs evict this CPU's stack memory in L3 to external memory, stale stack cacheline data go to external memory.
3. After this CPU pop stack, this CPU read stale stack data from external memory(at this time, C bit is still off) and return to wrong address and crash happened.

Is there a possible chance to do so?

This issue happens easily originally. After we add stack maintenance function in psci_do_pwrdown_sequence(), our system can pass overnight CPU power on/off stress test.

soby-mathew, 2018-03-21:
Yes, the situation you describe is possible if there are stale cache lines corresponding to current CPU lingering in the L3 cache which may cause corruption if they are evicted while the CPU is powering up.

The 2 solutions to this problem are:
*  do stack maintenance during power down or invalidate the stack before using the stack during power up.
*  Enable caches and enter coherency before using the stack.

Using solution 1 is the the most easiest to try although I would prefer invalidate the full stack during power up rather than the stack maintenance during power down. Could you try this and let me know if that works and submit a patch to that effect ?

The 2nd solution is the best for performance as we can avoid cache maintenance for stack altogther.  But this would require the assembly implementation of `enable_mmu_arch()` C function.

leon-chen-mtk, 2018-03-22:
Hi Soby,
Invalidating full stack by mva cache operation during CPU powering up do not take effect on L3 cache since at this time MMU is not enabled. So doing cache maintenance in power down sequence take effect.
I think rewriting enable_mmu_arch() in assembly is good, although that will need maintainer's support.

soby-mathew, 2018-03-22:
>Invalidating full stack by mva cache operation during CPU powering up do not take effect on L3 cache since at this time MMU is not enabled. So I think solution 1 only doing cache maintenance in power down sequence can fix the problem.

Yes, true. If done in the power down sequence, you would have to do this as the very final step before wfi (to ensure that stack is not used after the cache maintenance). The hook `psci_plat_pm_ops->pwr_domain_pwr_down_wfi` can be used for this.

>I think rewriting enable_mmu_arch() in assembly is good, although that will need maintainer's support.

OK. We will schedule a task to implement this.

leon-chen-mtk, 2018-03-24:
Got it, we will do stack maintenance in ```pwr_domain_pwr_down_wfi ``` instead of ```psci_do_pwrdown_sequence```.

soby-mathew, 2018-04-06:
Internal ref : [GENFW-2364](https://jira.arm.com/browse/GENFW-2364)

kenkuang, 2018-06-08:
We encounter the same issue exactly, and , found the solution: "invalidate the stack before using the stack during power up." work well.
refer to [pull request #1408](https://github.com/ARM-software/arm-trusted-firmware/pull/1408)
Since the other efficient solution "Enable caches and enter coherency before using the stack." coming soon, we will wait for patch    @jeenu-arm @soby-mathew 
thank you very much

jeenu-arm, 2018-06-13:
@kenkuang I've just pushed a patch series to fix this. Would you like to test with this patch and let us know how it goes?

kenkuang, 2018-06-15:
Hi, @jeenu-arm
 We merged and test it. so far so good.
thank you very much.

closed 2018-06-27 by dp-arm