ARM-software/tf-issues #571: qemu: MULTI_CONSOLE_API=0 causes build error

antonio-nino-diaz-arm, 2018-03-22:
When trying to compile the qemu binaries with `MULTI_CONSOLE_API=0` there are some build errors. I get the idea that https://github.com/ARM-software/arm-trusted-firmware/pull/1299 intended to maintain compatibility with the old API.

I guess that, if the plan is to deprecate the old system, the easiest way to fix this is to remove the compatibility code and don't bother with compatibility code. And maybe add a check to the platform makefile to stop the build if `MULTI_CONSOLE_API=0`.

CC: @michpappas

michpappas, 2018-03-23:
@antonio-nino-diaz-arm oops I thought I tested that case. The fix is trivial, but 
I am now considering additional changes based on @jwerner-chromium's comments on https://github.com/ARM-software/arm-trusted-firmware/pull/1299. In any case, I will provide a fix soon, thanks.

michpappas, 2018-03-23:
BTW, removing the old API altogether is currently not an option as MULTI_CONSOLE_API is not supported in AArch32

antonio-nino-diaz-arm, 2018-03-23:
Oh, right, I forgot about AArch32...

closed 2018-03-28 by dp-arm