ARM-software/tf-issues #573: MISRA: weakly linked functions being flagged as violations

vwadekar, 2018-03-29:
There are a lot of weakly defined functions throughout the code and MISRA scans flag these instances as violations of Rule 5.1, Rule 5.8 and Rule 8.6

vwadekar, 2018-03-29:
@danh-arm 

>>  We can potentially redefine these rules to allow weak function usag

Can you elaborate?

>> these rules will have to be excluded unless there's a major architectural change

Vendors would still have to fix these violations to achieve certifications. Are you saying ARM wont be fixing these issues in the upstream repo?

danh-arm, 2018-04-03:
> Can you elaborate?

Sure. As commented in the spreadsheet: "we can instead mandate exactly 1 (weak/strong) or 2 (1 weak + 1 strong) definition(s)". But we would have to rely on manual inspection to enforce that, as I don't think any static analzers support that interpretation.

> Vendors would still have to fix these violations to achieve certifications. Are you saying ARM wont be fixing these issues in the upstream repo?

MISRA certification doesn't require 100% compliance; it's possible to specify deviations from specific rules. I know that you're aiming for 100% complaince, but Arm does not plan to achieve this in the upstream repo. The spreadsheet indicates our specific intentions for each rule.