ARM-software/tf-issues #585: HiKey 960 documentation is outdated

antonio-nino-diaz-arm, 2018-04-30:
The documentation of this platform is outdated and it no longer works.

CC: @hzhuang1

mcarlini, 2018-05-15:
Arm Landing Team (@npoushin) has now been tasked of maintaining and refreshing HiKey960 platform port and related instructions.

antonio-nino-diaz-arm, 2018-05-15:
Fair enough. I guess that the contact in `maintainers.rst` has to change, then.

hzhuang1, 2018-05-16:
Which part is outdated in HiKey960 document?

antonio-nino-diaz-arm, 2018-05-16:
@hzhuang1, the part that says how to flash the images with `sudo fastboot flash`. I can only get the first two partitions to the board, the `fastboot` partition can't be copied, it tells me that it can't find the partition. I think that the partition table is incorrect. I've also tried with the instructions here with no luck: https://github.com/96boards-hikey/tools-images-hikey960/blob/b5ae2c13438eaab429aa584dec7080c181aeee87/build-from-source/README-ATF-UEFI-build-from-source.md

vchong, 2018-05-16:
@antonio-nino-diaz-arm How's your ptable image built? Rather than `prm_ptable.img`, try using one of the below that fits your system. Guessing it might be the aosp one you need.

```
ptable-aosp-32g.img
ptable-linux-32g.img
```

hzhuang1, 2018-05-16:
@antonio-nino-diaz-arm It seems that your partition table is wrong. Try to generate prm_ptable.img first.