ARM-software/tf-issues #587: disable_mmu_icache_el3 function in TF-A

ruchi393, 2018-05-01:
Hi,

My understanding is that the data caches are automatically invalidated on start-up and reset for A53 cores . Currently in ATF, data caches are disabled at exit from Bl2 in EL3 mode and enabled again in BL3.

I checked with ARM support and based on that my understanding is that after disable operation on data cache , explicit cleaning and invalidation of cache should be done.  It needs to be handled after disabling the data cache or before enabling the caches. However I don't see that happening either in the disable_mmu_icache code or enable_mmu code.

Can you help confirm the reasoning behind not doing invalidation of caches ? Shoudl platforms handle that in platform specific code ?

Regards,
Ruchika

danh-arm, 2018-05-01:
The RW memory used by a BL is invalidated in the early entrypoint code. For BL31, this is done here:
https://github.com/ARM-software/arm-trusted-firmware/blob/master/include/common/aarch64/el3_common_macros.S#L273

It's not necessary (or correct) to clean the caches at this point because nothing has changed in main memory since the caches were disabled, and the caches may contain stale data.

On CPU power down, caches are cleaned and invalidated in the CPU library before power off (pre-v8.2 CPUs only). For A53 this is done here:
https://github.com/ARM-software/arm-trusted-firmware/blob/master/lib/cpus/aarch64/cortex_a53.S#L228

ruchi393, 2018-05-01:
@danh-arm, the invalidation which is done in https://github.com/ARM-software/arm-trusted-firmware/blob/master/include/common/aarch64/el3_common_macros.S#L273 is only for the memory area being used by BL31. This would not ensure that complete data cache gets invalidated.

In my platforms flow, BL2 and BL31 get executed out of different memory locations. The memory where BL2 executes out form is later used by BL33. Here if the data cahes are not invalidated after disabling, I observe issues in BL33. Ideally even if cleanup is not required after BL2 as is our case, shouldn't we be invalidating complete cache after disabling it.

danh-arm, 2018-05-01:
Hi @ruchi393 

> The memory where BL2 executes out form is later used by BL33.

That sounds unusual as typically we expect BL2 memory to be secure and BL33 to only execute in and access non-secure memory. But you are free to make your own choice here.

> Ideally even if cleanup is not required after BL2 as is our case, shouldn't we be invalidating complete cache after disabling it.

That's wasted effort if the other memory is not used. In your case BL33 should take responsibility for invalidating the memory it's going to use,

achingupta, 2018-05-01:
> References

>    Hi [1]@ruchi393
> 
>      The memory where BL2 executes out form is later used by BL33.
> 
>    That sounds unusual as typically we expect BL2 memory to be secure and
>    BL33 to only execute in and access non-secure memory. But you are free
>    to make your own choice here.
> 
>      Ideally even if cleanup is not required after BL2 as is our case,
>      shouldn't we be invalidating complete cache after disabling it.
> 
>    That's wasted effort if the other memory is not used. In your case BL33
>    should take responsibility for invalidating the memory it's going to
>    use,

Thinking aloud..

This sounds like a case of mismatched memory attributes where an area of memory
is first marked secure and then non-secure while the secure cachelines are still
live. Normal world will not be able to invalidate these cache lines.

I am also wondering what guarantee is there that after invalidation, the data
will not be pulled right back into the caches. The secure world will have to
ensure that the data is zeroed and caches wiped out?

danh-arm, 2018-05-01:
> This sounds like a case of mismatched memory attributes where an area of memory is first marked secure and then non-secure while the secure cachelines are still live. Normal world will not be able to invalidate these cache lines.

If that's the case then BL2 platform code may as well map in its own memory as non-secure to avoid this problem, assuming that this memory is not protected by a TZC. But secure world platform code (BL2 or BL31) should zero out any secrets in this memory and caches before handing over to the normal world, for security reasons.

If the platform has some kind of dynamic TZC capability allowing the memory to switch from secure to non-secure, then whatever code that does the switch should take care of zeroing out the memory/caches.

closed 2018-05-17 by ruchi393