ARM-software/tf-issues #588: Add support for Renesas R-Car Gen3 SoCs

fisaksen, 2018-05-18:
Add arm-trusted-firmware Renesas Gen3 R-Car family of automotive embedded SoCs. See https://elinux.org/R-Car for details.
The current arm-trusted-firmare can be found here: 
- https://github.com/renesas-rcar/arm-trusted-firmware
- https://github.com/renesas-rcar/optee_os

antonio-nino-diaz-arm, 2018-05-18:
Hi, thanks for informing us of that port!

We are wondering if you have created this issue so that we can upstream this port, or if you intend to do it. Unfortunately, I've just done a quick check and the rebase isn't trivial, there are several conflicts with the master branch. We can't do this ourselves. There are also changes to common code that would need a closer look at.

However, if @renesas-rcar is interested in rebasing the port and creating a pull request (and becoming the maintainer) we'd be happy to review it and merge it if everything looks good and it follows the [contributing guidelines](https://github.com/ARM-software/arm-trusted-firmware/blob/master/contributing.rst).

fisaksen, 2018-05-18:
I will do this on behalf of renesas.
Thanks,
Frode

ldts, 2018-07-24:
@antonio-nino-diaz-arm I am taking over this task (I did the poplar support last year while with Linaro). One question, the current rcar ATF supports multiple SoC variants _and_ boards under the same "platform". In order to upstream this code -and as an acceptance criteria- do we need the code to be split accordingly (ie, per SoC variant: H (high end), M (mid), E(entry class)) or is it acceptable to decide at runtime (example below)

```
	prr = mmio_read_32(RCAR_PRR) & RCAR_PRODUCT_MASK;
	switch (prr) {
	case RCAR_PRODUCT_H3:
		bl2_system_cpg_init_h3();
		break;
	case RCAR_PRODUCT_M3:
		bl2_system_cpg_init_m3();
		break;
	case RCAR_PRODUCT_M3N:
		bl2_system_cpg_init_m3n();
		break;
	default:
		panic();
		break;
	}
```

antonio-nino-diaz-arm, 2018-07-24:
I guess this is really up to you. You can do it with different platforms, with different build options, or detected at runtime: https://github.com/ARM-software/arm-trusted-firmware/blob/bed42a5d28732aa6e933dc94c53a164ea5620d8e/plat/allwinner/common/sunxi_bl31_setup.c#L65-L97

antonio-nino-diaz-arm, 2018-07-24:
It's a matter of whether you can deal with the extra code used because of supporting more than one variant.

ldts, 2018-07-24:
sure I can deal with it - is just a bit confusing to external maintainers. But awesome: so ok to support multiple boards and SoC revisions from the same /plat/renesas/rcar-gen3 then?

antonio-nino-diaz-arm, 2018-07-24:
If that's how Renesas wants to build the firmware to make it easier to deploy the binaries I don't see why not. That's one of the objectives of having support for dynamic configuration: https://github.com/ARM-software/arm-trusted-firmware/blob/d48f193d11b4d4dca2675646ad76147f2d4765f2/docs/firmware-design.rst#dynamic-configuration-during-cold-boot

ldts, 2018-07-24:
reasons for not doing it this way is that it is a bit harder to read and maintain - it is easier to break SoC versions or boards that are not being developed... and it also increases the sizes of the different binaries. A reason in favor is that it decreases the size of the git repo. But the policy sounds good to me, will handle all under a single gen3 platform. thanks for the prompt replies!

antonio-nino-diaz-arm, 2018-07-24:
Yeah, it's a trade-off, that's why we leave that as a choice. There are upstreamed platforms that share code and others that don't, both are fine by us.

No problem!

danh-arm, 2018-07-26:
Hi @ldts. Do you have permission from Renesas to upstream this platform on their behalf? I'm a bit uncomfortable if this is done without their awareness/permission, in case it diverges from where they want it to go. It also makes the copyright ownership a bit messy. Regards, Dan.

ldts, 2018-07-26:
@danh-arm yes I understand your point and yes I do have permission. As I was commenting to @antonio-nino-diaz-arm on another channel, it would be very helpful to have some forum where some of the design decisions can be discussed early rather than late; for instance renesas's ATF does not implement a BL1 but instead performs the EL3 to EL1 transition as the BL2 entry point before continuing with BL2. In their git tree they modified the BL2 linker script to place a reset which I think it might be hard to maintain but with a small change [ https://github.com/BayLibre/arm-trusted-firmware/commit/df45ffc98116a661135b8d9a134f677754a3ecf6 ] we can maybe have a generic interface for the platforms what would like to do this instead of having a dedicated BL1?. would something like this be acceptable or is BL1 implementation mandatory for aarch64? what would be the best way to get clarification in areas like this?

danh-arm, 2018-07-26:
Hi @ldts

> yes I do have permission

OK, good.

> it would be very helpful to have some forum where some of the design decisions can be discussed early rather than late...

As Soby said, it's difficult for Arm to set up an IRC channel but this might be easier after the transition to trusted-firmware.org. However, it's possible to have a conference call to discuss this kind of thing. This can be arranged by email.

> for instance renesas's ATF does not implement a BL1 but instead performs the EL3 to EL1 transition as the BL2 entry point before continuing with BL2

It sounds like the Renesas proprietary Boot ROM expects the next firmware to execute at EL3, and they've created a platform specific solution for this. The latest upstream code has the `BL2_AT_EL3` feature, which should mean the Renesas solution is unnecessary. Search for this in the code and see what you think.

> ... is BL1 implementation mandatory for aarch64

No, we expect some vendors to have their own proprietary Boot ROM (although we'd like them to migrate to TF-A BL1). We have to support existing devices out there.

ldts, 2018-07-27:
@danh-arm thanks for the info. Using BL2_AT_EL3  does not seem straightforward since Renesas' BL2 executes at EL1 despite entering at EL3 (this way for example it uses SMC to BL1 to load the next image..); do you think that maybe allowing BL2 firmwares the option of switching exception levels before the stage entry point could be allowed as an ATF configuration option?

ldts, 2018-07-30:
@antonio-nino-diaz-arm @danh-arm I can boot using BL2_AT_EL3 as suggested. I still think the above (RESET_TO_BL2) might be useful to some but it is no longer blocking me.

danh-arm, 2018-07-30:
> I can boot using BL2_AT_EL3 as suggested. I still think the above (RESET_TO_BL2) might be useful to some but it is no longer blocking me.

Good to hear. As an alternative solution to BL2_AT_EL3, we considered wrapping BL2 with some code to drop from EL3 to S-EL1 as you suggest, but now BL2_AT_EL3 is there I don't see the value for both solutions. In any case, RESET_TO_BL2 would not be a good name since BL2 would not be the first BL to execute after reset (the proprietary ROM would be).

ldts, 2018-10-15:
https://github.com/ARM-software/arm-trusted-firmware/pull/1582
https://github.com/renesas-rcar/optee_os/pull/2