ARM-software/tf-issues #589: xilinx: Board zynqmp fails to compile with TSP enabled

antonio-nino-diaz-arm, 2018-05-18:
After the changes of https://github.com/ARM-software/arm-trusted-firmware/pull/1369, the following command line gives a linking error in BL32 (```undefined reference to `pm_get_chipid'```).

```
make CROSS_COMPILE=aarch64-linux-gnu- DEBUG=1 PLAT=zynqmp RESET_TO_BL31=1 SPD=tspd ZYNQMP_TSP_RAM_LOCATION=tdram all
```

CC: @sivadur

sivadur, 2018-05-25:
@antonio-nino-diaz-arm 
Thanks for finding out the issue.
I got what the issue is. BL32 is what we are not supporting fully. Let me see and get back to you on this.

jeenu-arm, 2018-06-20:
@sivadur any update on this?

We generally try to include build-only configs for all non-Arm platforms in our CI. Build options we use for zynqmp are:

```
	CROSS_COMPILE=aarch64-linux-gnu-
	PLAT=zynqmp
	RESET_TO_BL31=1
	SPD=tspd
```

and

```
	CROSS_COMPILE=aarch64-linux-gnu-
	PLAT=zynqmp
	RESET_TO_BL31=1
	SPD=tspd
	ZYNQMP_TSP_RAM_LOCATION=tdram
```

As @antonio-nino-diaz-arm noted earlier, the build for zynqmp fails, and our CI tests are flagged as failure. Although not ideal, we could disable the build for zynqmp, and we'd like to keep that as a last resort.

Could you submit a fix for the issue? Or do you have a different set of build options that you use to build the platform?

sivadur, 2018-06-20:
@jeenu-arm For now, you can remove SPD flag from build options. This should work.

BTW, i am looking for a build flag for BL32  in order to build my code conditional compilation as below. Could you please let me know if you are a aware of any such flag. 

#ifdef BUILD_BL32

#else 

#endif

closed 2018-06-21 by dp-arm