ARM-software/tf-issues #597: [RFC] Add MMC framework

Yann-lms, 2018-06-20:
I intend to push a new MMC framework, as an RFC for the moment, in drivers/mmc/mmc.c.

It is largely based on existing eMMC framework by Haojian Zhuang (@hzhuang1). 

The MMC framework supports both eMMC and SD card devices. It was written as a new framework since breaking few eMMC framework APIs. 

At card probe and after the reset to idle command (CMD0), a Send Interface Condition Command is sent (CMD8) to distinguish between eMMC and SD card devices. eMMC devices go through the same sequence as in the former eMMC framework. Else the framework uses commands dedicated to SD-cards for init or frequency switch.

A structure is created to share info with the driver.
It stores:
- the MMC type (eMMC, SD or SD HC)
- the device size
- the max frequency supported by the device
- the block size: 512 for eMMC and SD-HC and read from CSD structure for older SD-cards

Restriction to align buffers on block size has been removed.

The MMC framework includes some MISRA compliance coding style maybe not yet ported in the existing eMMC framework.

hzhuang1, 2018-06-23:
I hope there's only one framework. Could it work fine on hikey? If so, we could replace the emmc framework with this one?

Yann-lms, 2018-06-25:
@hzhuang1 That was the goal of this work. I cannot check on hikey, but I don't see any reason that it couldn't work. I first worked with your framework to manage eMMC. And as I wanted to support SD-cards, I made the updates in the new files.
As I said there are some API changes, the main is the new parameter for init function, to deal with the structure shared with framework and driver. You have to create a struct mmc_device_info in dw_mmc.c, before calling mmc_init.

I will push a new version for @robertovargas-arm comments.

robertovargas-arm, 2018-06-25:
The original framework was doing some alignment checks and invalidations, but they should be done in the low level drivers (or in the platform code). In the same way, the error checks weren't exhaustive and there were conditions (with defective hardware) where the code could hang. @hzhuang1, if @Yann-lms  attends all my comments then I think you will have to review your low level drivers to improve these points before using the new framework.

Yann-lms, 2018-06-28:
Hi, I've pushed a new version of the framework, after @robertovargas-arm review, and internal comments.
@hzhuang1 you may also have to deal with all the command responses, as we now have:
`#define MMC_RESPONSE_R(_x)		U(_x)`

I had another question: I wonder if I can change clk and bus_width to unsigned in mmc_init, mmc_enumerate and mmc_set_ios? This would require dw_mmc_params changes accordingly. And it will avoid some casts.

hzhuang1, 2018-06-29:
@Yann-lms I'm fine on your implementation. There's only one issue in mmc_write operation.

It seems that you move out the cache operation. Does it mean that you move all cache operation in the lower mmc device driver?

I'm fine on changing clk and bus_width to unsigned variable type.

hzhuang1, 2018-06-29:
@Yann-lms I'll be on holiday in the next two weeks. I may not give out comments on time. Basically, I think this mmc framework is good. But I don't have time to verify it on hikey yet.

Yann-lms, 2018-07-03:
@hzhuang1 I removed cache operations after @robertovargas-arm 's review, and those operations should now be done in driver and no more in framework. I will push a new patch today with the correction of the write function, and the change to unsigned for clk and bus_width.

closed 2018-07-13 by dp-arm