ARM-software/tf-issues #599: Question on PSCI cache maintenance

glneo, 2018-06-25:
Looking at the following line and its comment above:
https://github.com/ARM-software/arm-trusted-firmware/blob/master/lib/psci/psci_on.c#L81

If our local cache may contain stale data then we should invalidate it before the read, not flush it. Am I misunderstanding something here? It does not make sense to flush right before a read, we are the ones retrieving the data, whatever data was in our cache will still be read the same after this operation.

Has anyone seen this "theoretical edge case" described in the comment, because to me it looks like this actually introduces one instead.

Andrew

danh-arm, 2018-06-26:
In the code comment, it's not the local (running) CPU's caches that may be stale, it's the target CPU's caches. The flush ensures the running CPU's data is flushed to main memory. When the target CPU starts up, it will invalidate its own caches, so the correct data is pulled in from main memory.

I'm not sure if this edge case has been seen in practice or not.

glneo, 2018-06-26:
If it is the target CPU's caches that may be stale then as you say the target CPU will invalidate its own caches on startups. It is a couple lines later where we do the flush of the new info for the target CPU so it can read it like you are talking about:
https://github.com/ARM-software/arm-trusted-firmware/blob/f790cc0a9c492cf3615c82574e2c3f1ff8af0a3d/lib/psci/psci_on.c#L100 

That is not what this code I am referring to does, the code flushes the local CPU's cache of the taget CPU data before the local CPU reads right back that very same data.

The edge case this looks to be trying to solve is when the target CPU exits coherency and then the cache flush happens, which we (local CPU) will not see. This seems to be what the comment is describing "And so the cache may contain stale data for the target CPU." In which case the fix is to invalidate our cache of this data here, not flush it, so our view of the target CPU data is fresh before this read on the following line.

danh-arm, 2018-06-27:
Hmm, I see what you mean now. This change was introduced in https://github.com/ARM-software/arm-trusted-firmware/commit/71341d23668b1700cd50c0abd74073e4351acbd9. I'm going to let Soby comment on this.

soby-mathew, 2018-06-27:
> The edge case this looks to be trying to solve is when the target CPU exits coherency and then the cache flush happens, which we (local CPU) will not see. This seems to be what the comment is [...]

You have described the edge case correctly. To explain the fix, there are a couple of things to consider:

1. The per-CPU data structures in TF-A are cache line aligned and only the `owning` CPU is allowed to write to it and other `non-owning` CPUs are only allowed to read it. The CPU-ON case is the exception to this rule wherein the local CPU modifies the target CPU data and hence the need for the complex write sequence [here](https://github.com/ARM-software/arm-trusted-firmware/blob/6d769420b098f36717c99d4a6a2455fd6bd8fd23/lib/psci/psci_on.c#L94-L115).

2. PSCI_CPU_ON can be called at any point in time (not only after the target CPU has turned OFF). Which means that doing an invalidate by virtual address on target CPU data is wrong because it may invalidate any writes done by target CPU if it were within the power down sequence (i.e it is within psci_do_cpu_off()), but has not flushed out its cache lines yet.

So, it is wrong to invalidate target CPU's data from local CPU because the local CPU can never be sure of state of target CPU (whether it OFF/ON). Flushing (Clean + invalidate) is safe from local CPU because even if the target CPU data were `dirty`, it would be cleaned out to memory and the invalidate will ensure that the local CPU sees the updated value.

>Has anyone seen this "theoretical edge case" described in the comment, 

No, we haven't seen this issue in practice. But, as I mentioned above, flushing the target CPU data from local CPU is safe and will resolve the theoritical issue.

Hope that clarifies.

glneo, 2018-06-27:
This does not clarify the issue for me. There are 3 locations being discussed here, and they seem to be getting mixed up.

We have:
A. The local CPUs cached version of the target CPU data.
B. The target CPUs cached version of the target CPU data.
C. The target CPU data in shared memory.

> So, it is wrong to invalidate target CPU's data from local CPU because the local CPU can never be sure of state of target CPU (whether it OFF/ON).

The local CPU cannot invalidate the B, only its own A, right?

> Flushing (Clean + invalidate) is safe from local CPU because even if the target CPU data were dirty, it would be cleaned out to memory

If A is dirty because B has changed and written to C but the local CPU did not see this change, then flushing A will corrupt C.

> Which means that doing an invalidate by virtual address on target CPU data is wrong because it may invalidate any writes done by target CPU if it were within the power down sequence (i.e it is within psci_do_cpu_off()), but has not flushed out its cache lines yet.

Which target CPU data, A, B, or C? Invalidating A will not do anything to B or C.

soby-mathew, 2018-06-28:
> We have:
A. The local CPUs cached version of the target CPU data.
B. The target CPUs cached version of the target CPU data.
C. The target CPU data in shared memory.

> The local CPU cannot invalidate the B, only its own A, right?

No, the caches are coherent and can snoop into each other for most part of the code execution once the initial setup is done. Hence whilst in this `cache coherent` mode, any cache maintenance operation by virtual address will affect the data if present anywhere on the cache network. The DCIMVAC of `aff_info_state` will invalidate the cached copy on the cache network when `cache coherent`. Hence it is not safe to invalidate the same from local CPU.

> If A is dirty because B has changed and written to C but the local CPU did not see this change,
 then flushing A will corrupt C.

There are 2 statement here which I will break down to explain when target CPU has exited the `cache-coherency` and the `observability` domains :

*  A is dirty because B has changed and written to C but the local CPU did not see this change

A is not `dirty` but `stale` in this case because to be `dirty` it has to be written to by local CPU A. As I mentioned earlier, the target CPU data is never written to by the local CPU A upto this [point](https://github.com/ARM-software/arm-trusted-firmware/blob/6d769420b098f36717c99d4a6a2455fd6bd8fd23/lib/psci/psci_on.c#L81). This is something we ensure in the PSCI code.

* flushing A will corrupt C

No, because A is not `dirty` to begin with, flushing will just invalidates the cached `stale` copy and updated data is fetched from C. Hence flush is safe to be done at all times from local CPU A.

> Which target CPU data, A, B, or C? Invalidating A will not do anything to B or C.

Not correct as discussed above. Also, in addition to `aff_info_state`, the adjacent fields on the target [CPU data](https://github.com/ARM-software/arm-trusted-firmware/blob/73a9605197ba04aaf02d436a2a4ad56e695b426c/include/lib/el3_runtime/cpu_data.h#L85-L103) on that same line will be invalidated.

Even if target CPU B exits the `cache-coherency` (i.e snooping is disabled), the cache-maintenance operations done by the target CPU [here](https://github.com/ARM-software/arm-trusted-firmware/blob/6d769420b098f36717c99d4a6a2455fd6bd8fd23/lib/psci/psci_off.c#L139-L142) may still be visible to local CPU A (depends on the Cache coherent interconnect and other platform specific details). Hence the original issue may not manifest on many platforms.

glneo, 2018-06-28:
Yes, thanks @soby-mathew for making this more clear to me.

So if I got this right, then DCIMVAC on any core should cause the clean on every core's cache that belongs to the same coherency domain. What happens for a core outside of this, for instance on a different cluster on a non-cache-coherent interconnect? My guess would be the target core does not flush its cache but that this is okay as we would not have a stale copy of this data and the target will read from memory not its own local cache at that point.

soby-mathew, 2018-06-28:
> What happens for a core outside of this, for instance on a different cluster on a non-cache-coherent interconnect? 

If I understand your question, we don't support this configuration (I realized some of the comments in the code may suggest the contrary). If a CPU is always non-coherent and want to take part in the SMP system, all shared data needs to undergo software cache maintenance which is too hefty.
We assume all the CPUs within the SMP system are cache coherent via cache coherent interconnect during except during initial setup after reset and final stages of power down[1].

[1] DynamIQ CPUs are an exception as they are `cache-coherent` out of reset. That is why all the cache operations in PSCI can be NOP for DynamIQ.

glneo, 2018-06-28:
Okay, I think that clears things up for me, thanks again!

closed 2018-06-28 by glneo