ARM-software/tf-issues #603: optee_utils.c/tee_validate_header

nathan-menhorn, 2018-07-16:
The function:
static inline int tee_validate_header(optee_header_t *header)

does not check if the nb_images is greater than 0. If this function validates a header in which 0 is stored in header->nb_images. The function parse_optee_header will skip over the for loop at line 180:

     /* Parse OPTEE image */
     for (num = 0; num < header->nb_images; num++) {
         if (header->optee_image_list[num].image_id ==
                 OPTEE_PAGER_IMAGE_ID) {
             ret = parse_optee_image(pager_image_info,
                 &header->optee_image_list[num]);
         } else if (header->optee_image_list[num].image_id ==
                 OPTEE_PAGED_IMAGE_ID) {
             ret = parse_optee_image(paged_image_info,
                 &header->optee_image_list[num]);
         } else {
             ERROR("Parse optee image failed.\n");
             return -1;
         }
 
         if (ret != 0)
             return -1;
     }

and then cause the header_ep structure, header_ep->pc, header_ep->args.arg1, and header_ep_args.arg2, to be loaded with garbage data since header_ep is not initlizalized at the beginning of the parse_optee_header function.

Suggested update to tee_validate_header:

/** Invalid OPTEE header. */
#define OPTEE_HEADER_INVALID 0

/** Valid OP-TEE header. */
#define OPTEE_HEADER_VALID 1

 /*******************************************************************************
  * Check if it is a valid tee header
  * Return 1 if valid
  * Return 0 if invalid
  ******************************************************************************/
 static inline int tee_validate_header(optee_header_t *header)
 {
    /* Valid header image? */
    int valid = OPTEE_HEADER_INVALID;

    /* Check for a valid image. */
     if ((header->magic == TEE_MAGIC_NUM_OPTEE) &&
         (header->version == 2u) &&
         (header->nb_images > 0u)) {
         (header->nb_images <= OPTEE_MAX_IMAGE_NUM))
     {
         /* Indicate the image is valid. */
         valid = OPTEE_HEADER_VALID;
     }

     /* Otherwise, warn users about an unknown TEE. */
     else
     { 
         WARN("Not a known TEE, use default loading options.\n");
     }
     
     /* Return whether or the image is valid. */
     return valid;
 }

This update is also MISRA compliant.

nathan-menhorn, 2018-07-16:
Update the tee_valid_header function to invalidate OP-TEE headers where the number of binary images, nb_images, is equal to zero.

soby-mathew, 2018-07-17:
Thanks for reporting this issue. Could you also submit a pull request for the same ?

vchong, 2018-07-17:
OP-TEE probably won't generate a header with `nb_images = 0`, but maybe it's good to check anyway. Suggest below for consistency.

```
#define OPTEE_MIN_IMAGE_NUM		1
```
```
	if ((header->magic == TEE_MAGIC_NUM_OPTEE) &&
		(header->version == 2) &&
		(header->nb_images >= OPTEE_MIN_IMAGE_NUM) &&
		(header->nb_images <= OPTEE_MAX_IMAGE_NUM)) {
		return OPTEE_HEADER_VALID;
}
```

nathan-menhorn, 2018-07-17:
Will do.

Nathan Menhorn
Embedded Software Engineer
Core Vertical Markets | Xilinx, Inc.
Office: 505-798-6872
nathan.menhorn@xilinx.com<mailto:nathan.menhorn@xilinx.com> | www.xilinx.com<http://www.xilinx.com>

EXTERNAL EMAIL

nathan-menhorn, 2018-07-17:
Pull request created.

closed 2018-07-17 by nathan-menhorn