ARM-software/tf-issues #604: ARM PSCI interaction with PMIC

belladmv, 2018-07-17:
I have a question regarding the PMIC interface and controlling it.
We are working on a solution on a ARMv8 SoC without a dedicated core like Cortext-M0 for Power Management.

I am planning to interface PMIC to I2C and is supplying power to the ARM Core and other peripherals.

If the ARM Power needs to be turned off, then the PMIC needs to be controlled by the PSCI platform specific code. That essentially means having an I2C driver for the PMIC in the PSCI firmware implementation. However, for DVFS operations, which is not handled by the PSCI firmware, the Linux PM framework also needs to access the PMIC to set the voltage required. Any suggestion on how to handle this?

Thanks,
Mahantesh.

soby-mathew, 2018-07-17:
>If the ARM Power needs to be turned off, then the PMIC needs to be controlled by the PSCI platform specific code. That essentially means having an I2C driver for the PMIC in the PSCI firmware implementation. 

Yes, you would need to have an I2C driver for PMIC in PSCI platform specific code.

> However, for DVFS operations, which is not handled by the PSCI firmware, the Linux PM framework also needs to access the PMIC to set the voltage required. Any suggestion on how to handle this?

OK, 2 different software agents (PSCI and Linux) need to access the PMIC via I2C channel. Ideally in this case, you would have 2 dedicated coomunication channels to PMIC for the 2 agents and the channel for PSCI will be restricted to `secure-only` access.

From your description, it seems that you have a single I2C channel to PMIC. It is not recommended to share a single channel between a secure agent (PSCI) and linux. One solution is to restrict the channel to the secure world and the linux will have to request the secure world for DVFS via custom platform specific SMCs.

ldebieve, 2018-07-17:
Hi Soby, Could it be possible to use SCMI as generic entry point to manage this use case as DVFS was covered. It could be very useful to have a standard entry point (like PSCI) for SCMI if power management is made by TF-A and not by dedicated SCP. All part are already available in linux except SCMI/SMC maibox driver which is not yet integrated in linux.

soby-mathew, 2018-07-17:
As I understand, SCMI also assumes the presence of 2 different physical channels for DVFS by linux and power management by PSCI. The mailbox accessed by linux for DVFS would be different from the mailbox accessed by PSCI for SCP communication.  The problem of sharing channel between PSCI and linux is not solved by SCM I think.

soby-mathew, 2018-07-17:
On further thoughts, there was/is a possibility for SCMI to use SMCs as transport (instead of mailbox) but this is not currenly implemented as the overhead of using SMCs for performance/sensor management was considered to be expensive. This could be an option in this case, but this would need further clarification in the SCMI specification I think. I would need to confirm this.

belladmv, 2018-07-17:
To make it more complex, the PMIC could be a MFD with battery management support apart from power supply regulators.

soby-mathew, 2018-07-17:
> Soby, this was clearly my point (about scmi mailbox used as SMCs). I know that there is clearly a matter of overhead using this method.

Ah, sorry, I misunderstood. `Mailbox` is used in SCMI specification to refer to the underlying transport mechanism based on doorbell. Yes, using SMC as transport for SCMI is clearly the suitable solution but the spec is silent on this which needs clarification.

soby-mathew, 2018-07-18:
> Are you saying that there are PMICs with two I2C interfaces?

I was referring to platforms with dedicated power controller which have 2 separate channels for communication for the 2 security states.

hzhuang1, 2018-07-18:
>>Are you saying that there are PMICs with two I2C interfaces?

>I was referring to platforms with dedicated power controller which have 2 separate channels for communication for the 2 security states.

I think that it's too hard to find the dedicated power controller which have 2 separate channels for communication. Most of them are not produced any more, since they're dedicated to special usage.

As a workaround, they can add a microcontroller to handle the I2C communication in circuit. It should be easy to implement. Whether it's in secure mode or not, AP should always send the request to microcontroller. The microcontroller is the only one governor to controller regulators. If AP controls PMIC directly, two governors (PSCI and Linux) will cause confliction in some time.

belladmv, 2018-07-19:
Does it mean that most of the SoCs are going for a design with a dedicated micro controller, within/outside SoC, for handling Power Management functionality?

soby-mathew, 2018-07-21:
Yes, this is the case as far as I know.

belladmv, 2018-07-24:
Thanks Soby. 
In summery, there are two possibilities:
1> To use a dedicated micro controller for handling PM operations including DVFS and other PSCI functions. PMIC will be interfaced to external micro controller.
2> Handle DVFS via custom platform specific SMCs.

Please let me know if there are any other options possible without using micro controller.

soby-mathew, 2018-07-27:
As @ldebieve suggested, implementing the SMC transport for SCMI is a good solution in this case. SCMI is a generic standard [1] and there should ideally be generic support for this both in kernel and firmware.
The SMC transport would also ideally need clarification in the SCMI specification. 

As I understand, there are already some patches in this direction available for the kernel. Similar implementation has to be done for TF-A once the SCMI specification has been clarified. 

There is a performance penalty for using SMCs for DVFS but this is a platform choice. If the penalty is prohibitive, then the option 1 mentioned above needs to be done.

[1] http://infocenter.arm.com/help/topic/com.arm.doc.den0056a/DEN0056A_System_Control_and_Management_Interface.pdf

belladmv, 2018-07-29:
Thanks Soby. Could you please point me to the linux kernel patches implementing SCMI?