ARM-software/tf-issues #606: Storing ROTPK in RPI3

Christian56, 2018-07-22:
Hi, 
im having difficulties understanding how ARM Trusted Firmware works and a few questions regarding,  where the ROTPK resides and how it can be sure that it has not been tampered with. Im using OP-TEE on the rpi3.
 From my understanding  on these config params (https://github.com/OP-TEE/optee_os/blob/master/documentation/optee_with_auth_framework.md) I use a key pair which i generated as ROTPK and embedded a hash on bl1 and bl2 (https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/user-guide.rst). But when i compile the OS where is the ROTPK stored ? 
As stated in (https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/trusted-board-boot.rst) a SHA-256 hash is stored in trusted root-key storage registers, however based on the arm doku(http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0515b/CHDFCFHB.html) this is just for the juno board (?). 
Also from my understanding BL1 should be compiled when i compile OP-TEE (because it has the hash of ROTPK) so it shouldn't reside in a ROM(?). 
So does BL1 reside on the SD-Card and is protected by EL3 (https://github.com/ARM-software/arm-trusted-firmware/blob/master/docs/firmware-design.rst#12bl1) ?

antonio-nino-diaz-arm, 2018-07-23:
As the [documentation says](https://github.com/ARM-software/arm-trusted-firmware/blob/aa49bde8a3e8171fbe25751b4673624c80134f59/docs/plat/rpi3.rst), the Raspberry Pi 3 isn't a secure platform. This is just a port to experiment with. It's mainly useful because of how cheap and widespread it is. Also, it supports PSCI, so you can boot any OS with support for PSCI without needing cheats like spin tables (which is what they do right now). The documentation also explains how to do this.

However, there is no secure memory so, in practice, the normal world (EL2) can see all the memory of the system. Even if you hide some memory from EL1 by setting up stage 2 translation tables in EL2 to virtualize EL1, EL1 can just modify the contents of the SD card and reboot the system, which will load the new binaries.

In a regular platform, the root of trust is stored in ROM. However, the ROM of the RPi3 only contains a VideoCore bootloader that loads a payload for the VC from the SD, and that payload eventually loads from the SD card the final VC binary and the Arm payloads (armstub8.bin and kernel8.img). At no point there is any security involved: you can simply modify the binaries and reboot the system if that's what you want. All you need to reboot the system is to write a value to a register and setup the watchdog to reboot the system. The watchdog isn't secure so the normal world can use it.

Does this answer your question?

Christian56, 2018-07-23:
yes i guess it does. Thanks a lot for your detailed answer :)

closed 2018-07-23 by Christian56