ARM-software/tf-issues #607: IRC/Slack channels

ldts, 2018-07-25:
would it be possible to have a dedicated irc or slack channel to give visibility to design discussions as well as the platform up-streaming process?

soby-mathew, 2018-07-26:
TF-A will soon become a Linaro community project (see https://www.trustedfirmware.org/) . Currently this is a big legal/IT hurdle to do but this is something that can be setup then trustedfirmware.org materializes. Perhaps there will be a mailing list as well.

dp-arm, 2018-07-30:
@ldts We are looking into this, it may be possible to setup something sooner rather than later.

dp-arm, 2018-07-30:
@ldts btw the channel has been created

antonio-nino-diaz-arm, 2018-07-31:
Ironically, today we are having some network issues so we can't connect to freenode.

closed 2018-08-15 by dp-arm