ARM-software/tf-issues #608: platform_oid.h missing ?

Christian56, 2018-08-03:
Hi,
currently im trying to configure OP-TEE to boot in secure mode on my rpi3. However as im trying to compile the OS, I get the error that "platform_oid.h" is missing. It is included in tools/cert_create/src/cert.c and should be in the same directory. 
As I dont find platform_oid.h in the repo either. Im not sure how to continue. I dont have a tbbr_oid.h file either :( 
What am I missing ? Should it be created on build ? 
Sorry for the inconvenience :/

antonio-nino-diaz-arm, 2018-08-03:
Could you paste here the command line you are using to compile it?

Christian56, 2018-08-04:
CROSS_COMPILE=" ~/devel/optee2/build/../toolchains/aarch64/bin/aarch64-linux-gnu-" 
make -C ~/devel/optee2/build/../arm-trusted-firmware BL32=~/devel/optee2/build/../optee_os/out/arm/core/tee.bin DEBUG=1 V=0 CRASH_REPORTING=1 LOG_LEVEL=40 PLAT=rpi3 SPD=opteed all
make[1]: directory „~/devel/optee2/arm-trusted-firmware“ enter
Including services/spd/opteed/opteed.mk
Including common/auth/polarssl/polarssl.mk
Building rpi3
  CC      src/cert.c
src/cert.c:44:26: fatal error: platform_oid.h: file or directory not found
compilation terminated.
Makefile:85: the rule for the source „src/cert.o“ failed

antonio-nino-diaz-arm, 2018-08-04:
Are you using the upstream version of the firmware? Because that output is talking about ``common/auth/polarssl/polarssl.mk``, which was removed three years ago in commit https://github.com/ARM-software/arm-trusted-firmware/commit/962f7c512ae63129471506f94003902a3245d54a.

I think that the optee repositories haven't accepted yet the pull requests that enable support for this upstream port:

- https://github.com/OP-TEE/build/pull/269
- https://github.com/OP-TEE/manifest/pull/119

jbech-linaro, 2018-08-06:
@antonio-nino-diaz-arm you are right, with our current master branch from build/manifest you will get [a fork](https://github.com/OP-TEE/manifest/blob/master/rpi3.xml#L19) of Trusted Firmware that is very old. Once the PR's you are referring to are merged, this shouldn't be an issue for our RPi3 builds any longer. So for @Christian56, you either have to wait until those two PR's have been merged or use the patches from the PR's meanwhile waiting. With that being said, I think this ticket could be closed, since it's not an issue in upstream TF.

Christian56, 2018-08-06:
okay thanks a lot both of you :)

closed 2018-08-06 by Christian56