ARM-software/tf-issues #609: Missing barrier in bakery_lock_release?

raghuncstate, 2018-08-05:
I was reviewing the code for bakery lock and was wondering if we are missing a barrier in bakery_lock_release() function. In both the coherent and normal version, what prevents a store to normal memory(or to any memory not strongly ordered) inside the critical section, from being reordered after the "bakery->lock_data[me] = 0;" statement by the CPU? If the store in the critical section is ordered after the store to bakery->lock_data[me], but before the dsb() or write_cache_op(), it is possible that the store to the lock data structure is visible to other CPU's before the critical section store is completed. Is there something i'm missing?

jeenu-arm, 2018-08-08:
@raghuncstate Thanks for raising this issue. There do appear to be missing barriers, and as you say, it's likely that a contender will observe accesses in the critical section before the lock is released; same goes on the read side.

We have a patch in the works to fix this by fencing critical section code by inserting load and store barriers in the get and set functions, respectively.

raghuncstate, 2018-08-08:
You're welcome! and thanks for your response!