ARM-software/tf-issues #610: MULTI_CONSOLE_API and pl011 driver for v1.5

apm-tjagad, 2018-08-08:
Hi,

I'm trying to add support for MULTI_CONSOLE_API on our aarch64 platform. We are using ATF v1.5 tag as the base of our port.

Our platform uses the pl011_console driver for console related functionality. Our current port is working fine and boots to the Linux promt when MULTI_CONSOLE_API is disabled.

When MULTI_CONSOLE_API is enabled, We find that code freezes in early stages in BL1 probably when console_pl011_register is called. It seems that line 105 (code snippet below from drivers/arm/pl011/aarch64/pl011_console.S) seems to be causing an issue when it is trying to store the base address of the UART in the console_pl011_t structure. 
```
101 func console_pl011_register
102         mov     x7, x30
103         mov     x6, x3
104         cbz     x6, register_fail
105         str     x0, [x6, #CONSOLE_T_PL011_BASE]
106 
107         bl      console_pl011_core_init
108         cbz     x0, register_fail
109 
110         mov     x0, x6
111         mov     x30, x7
112         finish_console_register pl011
```
As per our experiments if that access is replaced with functionality similar to console_init  (from drivers/console/aarch64/deprecated_console.S) shown below  it works to some extent. 
```
 43 func console_init
 44         /* Check the input base address */
 45         cbz     x0, init_fail
 46         adrp    x3, console_base
 47         str     x0, [x3, :lo12:console_base]
 48         b       console_core_init
 49 init_fail:
 50         ret
 51 endfunc console_init
```
Since we are using this code which in more or less way does not use the framework provided by MULTI_CONSOLE_API it works but fails going forward.

I'm kind of clueless on what code needs to be used in assembly so that we can avoid the crash and in turn able to use MULTI_CONSOLE_API.

Any pointers to solve the issue would be helpful.

Thanks and Regards,
Tushar

antonio-nino-diaz-arm, 2018-08-08:
So you are using the upstream PL011 driver, right? It looks to me like something else is going on. Have you taken a look at the code of other platforms to see how they do it and compare? ``plat/arm/common/arm_console.c`` for example. There may be a problem with the way you are reserving the structures to register the console.

amp-tjagad, 2018-08-08:
Hi Antonio Niño Díaz,

Thanks for your reply.

Yes, we're using the upstream PL011 driver on our platform.

Kindly note that we're using the tag v1.5 wherein the the file plat/arm/common/arm_console.c has been added later on. Our changes have been based on plat/qemu for MULTI_CONSOLE_API. We have checked the code for master:plat/arm/common/arm_console.c and have found that it is similar to what we've in our code.

We checked if some fixes have been done post v1.5 to drivers/console/ and drivers/arm/pl011/ as well. We didn't find any significant difference between the two versions.

Also, the memory used for the structures referenced while registering the console are allocated in a similar way as plat/qemu and plat/arm/common/arm_console.c. We have used static keyword to ensure the memory is allocated on the bss. I'll double check the map file and verify the address for the same.

Should the address of the static structure be loaded using adrp like in the original code before updating the members ? The code for console_register is exactly doing the same to update the root of the linked list ?
```
 44 func console_register
 45 #if ENABLE_ASSERTIONS
 46         cmp     x0, #0
 47         ASM_ASSERT(ne)
 48         adrp    x1, __STACKS_START__
 49         add     x1, x1, :lo12:__STACKS_START__
 50         cmp     x0, x1
 51         b.lo    not_on_stack
 52         adrp    x1, __STACKS_END__
 53         add     x1, x1, :lo12:__STACKS_END__
 54         cmp     x0, x1
 55         ASM_ASSERT(hs)
 56 not_on_stack:
 57 #endif /* ENABLE_ASSERTIONS */
 58         adrp    x14, console_list
 59         ldr     x1, [x14, :lo12:console_list]   /* X1 = first struct in list */
 60         str     x0, [x14, :lo12:console_list]   /* list head = new console */
 61         str     x1, [x0, #CONSOLE_T_NEXT]       /* new console next ptr = X1 */
 62         mov     x0, #1
 63         ret
 64 endfunc console_register
```
Thanks and Regards,
Tushar

antonio-nino-diaz-arm, 2018-08-08:
The initial console registration happens with the MMU disabled so this can't be because of trying to access a RO page. Have you checked that the pointer to the struct is aligned to 64 bits? That's the only reason that I can think of that could make a write fail with MMU disabled.

amp-tjagad, 2018-08-08:
Hi Antonio Niño Díaz,

Thanks for your reply and help in narrowing this down.

I don't have access to the platform right now to check the address for the structure. I'll get back on this front. 

On this note other platforms don't do anything to align the static structure to a 64-bit address.

Kindly find the code snippet from master:plat/arm/common/arm_console.c
```
 16 #if MULTI_CONSOLE_API
 17 static console_pl011_t arm_boot_console;
 18 static console_pl011_t arm_runtime_console;
 19 #endif
```

Is there some thing missing which I need to look at on this front ?

Thanks and Regards,
Tushar

antonio-nino-diaz-arm, 2018-08-08:
Hi,

It is very unlikely that the problem is a misaligned access. The fields of the struct are 64-bit wide, so it will naturally be aligned to 64 bit. It is worth checking, though, we've had similar issues in the past.

The only two reasons I can see for this crash is that the address of the struct isn't aligned because it somehow gets corrupted or because there is some weird behaviour in your compiler. Are you using the aarch64 compiler from the Linaro 17.10 release or a different one?

amp-tjagad, 2018-08-08:
Hi Antonio Niño Díaz,

We're using the compiler optimized for Ampere Computing for their SOC's.

I'll check if I can find some information from the compiler team.

Thanks and Regards,
Tushar

amp-tjagad, 2018-08-09:
Hi Antonio Niño Díaz,

I tried checking the behavior with both our compiler and Linaro compiler today. In both cases I fail to get MULTI_CONSOLE_API working. Kindly find the snippet of the map file below from both compilers. In both cases the addresses are 64-bit aligned.

**Using Our Compiler**
```
2500  .bss.xgene_console
2501                 0x000000001d071d90       0x30 ./build/xgene/release/bl1/osprey_console.o
```

**Using Linaro Compiler**
```
2492  .bss.xgene_console
2493                 0x000000001d071d80       0x30 ./build/xgene/release/bl1/osprey_console.o
```

Please note that I turned off MULTI_CONSOLE_API with Linaro compiler and found that it worked with Linaro toolchain as well. 

The Linaro toolchain which I'm using is https://releases.linaro.org/components/toolchain/binaries/latest/aarch64-linux-gnu/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu.tar.xz

Thanks and Regards,
Tushar

antonio-nino-diaz-arm, 2018-08-09:
Ok, so the alignment is correct.

How do you know it crashes at that instruction, have you used a debugger or is it just an educated guess?

I suppose that you have working ``plat_crash_console_init`` and ``plat_crash_console_putc``. I'm out of ideas, so I think that you should just try to call ``plat_crash_console_putc`` at different points during ``console_register`` and ``console_pl011_register`` and see where it crashes.

antonio-nino-diaz-arm, 2018-08-09:
Just to make sure that this is not the case... Are your console driver functions in assembly?

amp-tjagad, 2018-08-09:
Hi Antonio Niño Díaz,

No I've not yet tried the debugger yet. 

It is kind of the educated guess as one of my colleague replicated the code from `console_init (drivers/arm/pl011/aarch64/deprecated_console.S)` into the `console_pl011_register`.  Instead of populating to the `console_pl011_t->base` member of the static structure and the code worked to some extent. But since the MULTI_CONSOLE_API frameworks loops through the nodes of the linked list it fails at a later stage which is quite obvious due to the nature of the change.

I've implemented the `plat_crash_console_init` and `plat_crash_console_putc` and tried it once but the behavior was still the same. So I gave up on the effort of trying it. I will verify this and try to use it till the time I get the JTAG going.

By console driver functions, I hope you mean the calls to the wrapper functions used to register the console. No they are C functions. Kindly find the code snippet of the same below

```
 34 #if MULTI_CONSOLE_API
 35 static console_pl011_t console;
 36 
 37 int console_pl011_core_putc(int c, uintptr_t base_addr);
 38 #endif /* MULTI_CONSOLE_API */

 73 void xgene_console_init(void)
 74 {
 75         uint32_t uart_base;
 76         uint32_t uart_clk;
 77 
 78         uart_base = get_uart_base();
 79         uart_clk = xgene_get_uart_clk();
 80 
 81 #if MULTI_CONSOLE_API
 82         (void)console_pl011_register(uart_base, uart_clk,
 83                         XGENE_BOOT_UART_BAUDRATE, &console);
 84 
 85         console_set_scope(&console.console, CONSOLE_FLAG_BOOT |
 86                         CONSOLE_FLAG_RUNTIME);
 87 #else
 88         console_init(uart_base, uart_clk, XGENE_BOOT_UART_BAUDRATE);
 89 #endif /* MULTI_CONSOLE_API */
 90 }
```
Thanks for your time in help on the matter.

Regards,
Tushar

antonio-nino-diaz-arm, 2018-08-09:
Ok, everything looks fine. I don't know, I think you will have to use the debugger and see what is actually going on.

amp-tjagad, 2018-08-09:
Hi Antonio Niño Díaz,

I disabled MULTI_CONSOLE_API and added a `panic()` call in bl1_early_platform_setup is see the PC being dumped on the UART. Thus this verifies that the plat_crash_console_init and plat_crash_console_putc functions are working fine.

I enabled MULTI_CONSOLE_API and tried the same experiment but this time I don't see any thing on the console.

```
186 void bl1_early_platform_setup(void)
187 {
188         panic();
189 
190         xgene_bl1_early_platform_setup();
191 }
```
This is a strange as the console registration is done as a part of the `xgene_bl1_early_platform_setup()` function and it should have worked.

I've tried checking the bss, data and text segment but haven't found any overlaps and conflicting areas with and without MULTI_CONSOLE_API.

Thanks and Regards,
Tushar

antonio-nino-diaz-arm, 2018-08-09:
Well, your console is working and your code looks fine. I think that there must be something else going on, so it's better if we wait until you can use the debugger, as you said before.

antonio-nino-diaz-arm, 2018-10-04:
@amp-tjagad did you manage to fix this? Can we close this issue?

jwerner-chromium, 2018-10-04:
Note that the console_base pointer in the old console framework was intentionally allocated in the .data segment, not in the BSS, to work around some problem during early init. See this comment in deprecated_console.S:

        /*                                                                       
         *  The console base is in the data section and not in .bss              
         *  even though it is zero-init. In particular, this allows              
         *  the console functions to start using this variable before            
         *  the runtime memory is initialized for images which do not            
         *  need to copy the .data section from ROM to RAM.                      
         */

multi_console.S does the same thing for the global variables stored in the framework itself, but since it also uses a console struct that is passed in by the caller, the caller is responsible for where that struct is allocated. Maybe your issue is that you should be putting your console struct into the .data section as well. Try to see if this helps:

static console_pl011_t console __attribute__((section(".data")));

antonio-nino-diaz-arm, 2018-10-05:
@jwerner-chromium That's interesting, I guess that most platforms get away with it because they initialize the console in ``early_platform_setup`` after the C runtime is setup. At that point BSS is already cleared so there is no difference. As a matter of fact, you can't initialize it before without modifying common code. I guess you can register it from ``platform_mem_init``, but that's an assembly function so it's unlikely that anyone is doing that.

kuvivek81, 2018-10-08:
It worked when I did following changes in drivers/arm/pl011/aarch64/pl011_console.S
A persistent (Console_base) section is defined
.section .data.console_base ; .align 3
               console_base: .quad 0x0

And inside function console_pl011_register modified it like below:
func console_pl011_register
#if 0     //Original Code
	mov	x7, x30
	mov	x6, x3
	cbz	x6, register_fail
	adrp    x6, console_base	//PC relative offset
	str	x0, [x6, #CONSOLE_T_PL011_BASE]

	bl	console_pl011_core_init
	cbz	x0, register_fail

	mov	x0, x6
	mov	x30, x7
#else      //Modified Code
        mov     x7, x30
        adrp    x6, console_base        
	str 	x3, [x6, :lo12:console_base]
	ldr 	x6, [x6, :lo12:console_base]
        str     x0, [x6, #CONSOLE_T_PL011_BASE]

        bl      console_pl011_core_init
        cbz     x0, register_fail

        mov     x0, x6
        mov     x30, x7
#endif
	finish_console_register pl011

I think aligned address generation was not happening with the original instructions and memory access couldn’t be successfully completed and perhaps abort signals are asserted. 
Your thoughts please!

register_fail:
	ret	x7
endfunc console_pl011_register

antonio-nino-diaz-arm, 2018-10-08:
But now you are replacing the linked list by a single console, right? That's what console_base is doing, as far as I can see.

kuvivek81, 2018-10-08:
No I don't think so. 
X0 value is preserved and the function console_register from  drivers\console\aarch64\multi_console.S which is creating the list for multi-console is still intact.

antonio-nino-diaz-arm, 2018-10-08:
Oh, my bad, it's just hard to read with the format of your comment.

Then why do you need ```console_base```? You're only saving a value there, you don't actually use it afterwards. That's only used in the deprecated console driver.

kuvivek81, 2018-10-08:
Yes, right I don't need console_base later. It was used only as temporary resort to be used with adrp. I am not an assembly expert...if there's a better way to make it inline, let me know!

antonio-nino-diaz-arm, 2018-10-08:
The thing is that, as far as I can see, you are only removing the check: ``cbz x6, register_fail``

The rest of the code is essentially the same.

kuvivek81, 2018-10-08:
The removal of cbz x6, register_fail does not impact...I can include it, it was just a miss here. With below modification however:

adrp x6, console_base
str x3, [x6, :lo12:console_base]
ldr x6, [x6, :lo12:console_base]

I could see it started working.
What I feel is earlier the aligned memory access was not happening and the memory controller was asserting the abort signal.
With the use of adrp instruction, it is fixing the alignment issue and thus I am able to see it working.

antonio-nino-diaz-arm, 2018-10-08:
But that code is useless. ``console_base`` isn't used anywhere else. Your code is basically ``mov x6,x3``.

antonio-nino-diaz-arm, 2018-10-08:
I mean, you are doing:
```
console_base = x3;
x6 = console_base;
```

kuvivek81, 2018-10-08:
Hmm. 
My purpose was to align the address. 
As said, I am a beginner in assembly.  Agree that there could be a better way of doing it... unfortunately I could manage with this way only.

antonio-nino-diaz-arm, 2018-10-08:
But what address do you need to align? You aren't aligning anything like that.

kuvivek81, 2018-10-08:
I thought adrp is helping in forming a 4kb aligned page address.
And with the change I could see Multi Console started working for me, without which it wasn't.

kuvivek81, 2018-10-08:
Adrp here is used in conjunction with load and store with a 12 bit immediate offset.
Referring here: 
http://www.rowleydownload.co.uk/arm/documentation/gnu/as/AArch64_002dRelocations.html

antonio-nino-diaz-arm, 2018-10-08:
This
```
ldr x6,console_base
ldr x6,[x6]
```
and this
```
adrp x6, console_base
ldr x6, [x6, :lo12:console_base]
```
do the same read from memory. So I don't see how this can fix anything.

antonio-nino-diaz-arm, 2018-10-09:
I still think that your code isn't doing anything different than the original one. You are just saving x3 to ``console_base``, but you don't use that variable anywhere, so in practice you are just moving the value from x3 to x6 like in the original code. The only difference is that you have removed the check.

kuvivek81, 2018-10-09:
Do you think removal of check is making any difference? It can be added...it was just a miss and is for fail handling.

Yes, X3 is being stored in "console_base" and "console_base" is subsequently loaded to be pointed by X6.
This brings a big difference when X3 is directly moved to X6 in original case whereas X6 is loaded to point console_base.

With original code, booting halts in the beginning stage itself whereas it successfully completes when this  modification is induced.

antonio-nino-diaz-arm, 2018-10-09:
Yes, I think that the problem is the check. Could you try adding it back?

kuvivek81, 2018-10-09:
No that's not a problem. It's just a check.
I tried below modified code (check is introduced), and it's just working fine:

/* Below is defined to be used with modified code*/
.section .data.console_base ; .align 3
console_base: .quad 0x0
/***********************/

func console_pl011_register
#if 0
/* Original Code */

    mov     x7, x30
    mov     x6, x3
    cbz     x6, register_fail
    str     x0, [x6, #CONSOLE_T_PL011_BASE]
    bl      console_pl011_core_init
    cbz     x0, register_fail
    mov     x0, x6
    mov     x30, x7
#else
/* Modified Code */

    mov     x7, x30
    adrp    x6, console_base        
    str     x3, [x6, :lo12:console_base]
    cbz     x6, register_fail
    ldr     x6, [x6, :lo12:console_base]
    cbz     x6, register_fail
    str     x0, [x6, #CONSOLE_T_PL011_BASE]
    bl      console_pl011_core_init
    cbz     x0, register_fail
    mov     x0, x6
    mov     x30, x7
#endif

finish_console_register pl011
register_fail:

ret	x7

antonio-nino-diaz-arm, 2018-10-09:
I'm sorry, but I can't see how that change could fix anything related to that code. I think that maybe the write and read are acting as some sort of memory barrier or delay that hides a different problem.