ARM-software/tf-issues #612: Marvell a80x0 build broken?

jeenu-arm, 2018-08-14:
The steps I tried are:

```
git clone https://github.com/ARM-software/arm-trusted-firmware
git clone https://github.com/MarvellEmbeddedProcessors/mv-ddr-marvell
git clone https://github.com/MarvellEmbeddedProcessors/ble-marvell

pushd arm-trusted-firmware
make CROSS_COMPILE=aarch64-linux-gnu- PLAT=a80x0 SCP_BL2=/dev/null  \
  DEBUG=1 BLE_PATH=../ble-marvell MV_DDR_PATH=../mv-ddr-marvell V=1 fiptool all
```

But the build fails complaining about missing ``mv_ddr_if.h`` file. Are there specific branches of the external repositories that one's supposed to use?

@kostapr, please advise.

kostapr, 2018-08-14:
For DDR sources, please use "mainline" branch.
git clone https://github.com/MarvellEmbeddedProcessors/mv-ddr-marvell -b mv_ddr-armada-atf-mainline
Or if you already cloned it,  just "git checkout mv_ddr-armada-atf-mainline"

jeenu-arm, 2018-08-14:
OK, I tried with that branch, but link fails saying can't find ``printf``.

kostapr, 2018-08-14:
Do you plan to run Linux kernel tests in CI or just a build tests?

jeenu-arm, 2018-08-14:
Currently we only test that upstream platforms build successfully; no tests are run.

kostapr, 2018-08-14:
OK, this one should fix the broken build:
https://github.com/MarvellEmbeddedProcessors/ble-marvell/commit/4dd59a313b1934d6ec8fd8ae11b11275b8ae5edc

Please use the "atf-mainline" branch on MarvellEmbeddedProcessors/ble-marvell.git

jeenu-arm, 2018-08-14:
OK, thanks - it works now. However, I noticed that the dependencies are probably broken, and so builds with ``-j 8`` fails. But don't worry about this right now - the build system has other long-standing problems.

closed 2018-08-14 by jeenu-arm